MIT-SPARK/Kimera-VIO-ROS #118: Kimera_vio failed in catkin build | kimera_vio:make /home/kca/catkin_ws/logs/kimera_vio/build.make.003.log

kcaisfather, 2020-11-27:
Hello,

I tried to compile Kimera VIO for our project on ubuntu 18.04. I have a problem catkin build part. There weren't any problem before coming to this stage. 

Can you help me?

`___________________________________________________________________________
Errors     << kimera_vio:make /home/kca/catkin_ws/logs/kimera_vio/build.make.003.log                                                                                                                                
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/Histogram.cpp: In copy constructor ‘VIO::Histogram::Histogram(const VIO::Histogram&)’:
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/Histogram.cpp:80:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t i = 0; i < dims_; i++) {
                      ~~^~~~~~~
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/Histogram.cpp:86:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t i = 0; i < dims_; i++) {
                      ~~^~~~~~~
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/Histogram.cpp:90:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t k = 0; k < dims_; k++) {
                      ~~^~~~~~~
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp: In static member function ‘static void VIO::UtilsOpenCV::PlainMatchTemplate(cv::Mat, cv::Mat, cv::Mat&)’:
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp:446:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t i = 0; i < result_rows; i++) {
                      ~~^~~~~~~~~~~~~
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp:447:26: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (size_t j = 0; j < result_cols; j++) {
                        ~~^~~~~~~~~~~~~
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp: In static member function ‘static cv::Mat VIO::UtilsOpenCV::DrawCornersMatches(const cv::Mat&, const std::vector<cv::Point_<float> >&, const cv::Mat&, const std::vector<cv::Point_<float> >&, const std::vector<cv::DMatch>&, bool)’:
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp:601:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < matches.size(); i++) {
                   ~~^~~~~~~~~~~~~~~~
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp: In static member function ‘static void VIO::UtilsOpenCV::DrawCornersMatchesOneByOne(cv::Mat, const std::vector<cv::Point_<float> >&, cv::Mat, const std::vector<cv::Point_<float> >&, const std::vector<cv::DMatch>&)’:
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp:689:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < matches.size(); i++) {
                   ~~^~~~~~~~~~~~~~~~
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp: In static member function ‘static double VIO::UtilsOpenCV::MaxAbsValue(gtsam::Matrix)’:
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp:704:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t i = 0; i < M.rows(); i++) {
                      ~~^~~~~~~~~~
/home/kca/catkin_ws/src/Kimera-VIO/src/utils/UtilsOpenCV.cpp:705:26: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (size_t j = 0; j < M.cols(); j++) {
                        ~~^~~~~~~~~~
c++: internal compiler error: Süreç durduruldu (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/testKimeraVIO.dir/tests/testPipeline.cpp.o] Error 4
make[2]: *** Bitmemiş işler için bekliyor....
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
make: *** [all] Error 2
cd /home/kca/catkin_ws/build/kimera_vio; catkin build --get-env kimera_vio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
....................................................................................................................................................................................................................
Failed     << kimera_vio:make                              [ Exited with code 2 ]                                                                                                                                   
Failed    <<< kimera_vio                                   [ 1 minute and 14.4 seconds ]                                                                                                                            
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]                                                                                                                                 
[build] Summary: 14 of 16 packages succeeded.                                                                                                                                                                       
[build]   Ignored:   None.                                                                                                                                                                                          
[build]   Warnings:  None.                                                                                                                                                                                          
[build]   Abandoned: 1 packages were abandoned.                                                                                                                                                                     
[build]   Failed:    1 packages failed.                                                                                                                                                                             
[build] Runtime: 1 minute and 18.0 seconds total.           `

kcaisfather, 2020-11-27:
It was solved to resave histogram.cpp. We didnt do anything to change any file.

closed 2020-11-27 by kcaisfather

AlonSpinner, 2022-02-06:
I am having the exact same problem. 
resaved histogram.cpp, but still occuring:

```
/home/alon18/repos/kimera_ws/src/Kimera-VIO/src/utils/Histogram.cpp: In copy constructor ‘VIO::Histogram::Histogram(const VIO::Histogram&)’:
/home/alon18/repos/kimera_ws/src/Kimera-VIO/src/utils/Histogram.cpp:79:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t i = 0; i < dims_; i++) {
                      ~~^~~~~~~
/home/alon18/repos/kimera_ws/src/Kimera-VIO/src/utils/Histogram.cpp:85:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t i = 0; i < dims_; i++) {
                      ~~^~~~~~~
/home/alon18/repos/kimera_ws/src/Kimera-VIO/src/utils/Histogram.cpp:89:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t k = 0; k < dims_; k++) {
                      ~~^~~~~~~
In file included from /home/alon18/repos/kimera_ws/src/Kimera-VIO/src/loopclosure/LoopClosureDetectorParams.cpp:15:0:
/home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetectorParams.h: In constructor ‘VIO::LoopClosureDetectorParams::LoopClosureDetectorParams(int, int, double, cv::Point2d, bool, float, int, int, int, float, int, int, int, int, VIO::GeomVerifOption, int, int, double, double, bool, double, VIO::PoseRecoveryOption, int, double, double, bool, double, bool, bool, double, int, int, float, int, int, int, int, int, int, int, double, double, double, double)’:
/home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetectorParams.h:224:7: warning: ‘VIO::LoopClosureDetectorParams::fast_threshold_’ will be initialized after [-Wreorder]
   int fast_threshold_;
       ^~~~~~~~~~~~~~~
/home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetectorParams.h:199:10: warning:   ‘double VIO::LoopClosureDetectorParams::betweenRotationPrecision_’ [-Wreorder]
   double betweenRotationPrecision_;
          ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/alon18/repos/kimera_ws/src/Kimera-VIO/src/loopclosure/LoopClosureDetectorParams.cpp:19:1: warning:   when initialized here [-Wreorder]
 LoopClosureDetectorParams::LoopClosureDetectorParams(
 ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/alon18/repos/kimera_ws/src/Kimera-VIO/src/backend/VioBackendParams.cpp: In static member function ‘static void VIO::BackendParams::setIsam2Params(const VIO::BackendParams&, gtsam::ISAM2Params*)’:
/home/alon18/repos/kimera_ws/src/Kimera-VIO/src/backend/VioBackendParams.cpp:46:15: error: ‘struct gtsam::ISAM2Params’ has no member named ‘setCacheLinearizedFactors’; did you mean ‘cacheLinearizedFactors’?
   isam_param->setCacheLinearizedFactors(true);
               ^~~~~~~~~~~~~~~~~~~~~~~~~
               cacheLinearizedFactors
/home/alon18/repos/kimera_ws/src/Kimera-VIO/src/backend/VioBackendParams.cpp:51:15: error: ‘struct gtsam::ISAM2Params’ has no member named ‘setEvaluateNonlinearError’; did you mean ‘evaluateNonlinearError’?
   isam_param->setEvaluateNonlinearError(false);  // only for debugging
               ^~~~~~~~~~~~~~~~~~~~~~~~~
               evaluateNonlinearError
In file included from /home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/imu-frontend/ImuFrontend-definitions.h:26:0,
                 from /home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/imu-frontend/ImuFrontend.h:33,
                 from /home/alon18/repos/kimera_ws/src/Kimera-VIO/src/imu-frontend/ImuFrontend.cpp:16:
/home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
 static void getValidKeypointsFromStatusKeypointsCV(
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/imu-frontend/ImuFrontend-definitions.h:26:0,
                 from /home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/imu-frontend/ImuFrontendParams.h:19,
                 from /home/alon18/repos/kimera_ws/src/Kimera-VIO/src/imu-frontend/ImuFrontendParams.cpp:15:
/home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
 static void getValidKeypointsFromStatusKeypointsCV(
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector-definitions.h:28:0,
                 from /home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetectorParams.h:24,
                 from /home/alon18/repos/kimera_ws/src/Kimera-VIO/src/loopclosure/LoopClosureDetectorParams.cpp:15:
/home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/common/vio_types.h: At global scope:
/home/alon18/repos/kimera_ws/src/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
 static void getValidKeypointsFromStatusKeypointsCV(
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
[build]   Ignored:   None.                                                                                                                
[build]   Warnings:  None.                                                                                                                
[build]   Abandoned: 1 packages were abandoned.                                                                                           
[build]   Failed:    1 packages failed.                                                                                                   
[build] Runtime: 55.1 seconds total.            
```