MIT-SPARK/Kimera-VIO-ROS #122: kimera_rpgo build fail

IbrahimKaramavus1773, 2020-12-30:
**Description:**
Dear developers,
I am trying to build Kimera_VIO_ROS but i faced an error in building Kimera_RPGO. I am using Ubuntu 18.04 and ROS Melodic.The error is shown belown.

Any ideas how to fix it ?

**Command:**
```
# replace this line with the command(s) you used
```

**Console output:**
```

Errors     << kimera_rpgo:make /home/ibrahimkaramavus/catkin_ws/logs/kimera_rpgo/build.make.004.log                                                                                  
In file included from /home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/include/KimeraRPGO/GenericSolver.h:16:0,
                 from /home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:9:
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/include/KimeraRPGO/SolverParams.h: In constructor ‘KimeraRPGO::RobustSolverParams::RobustSolverParams()’:
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/include/KimeraRPGO/SolverParams.h:133:8: warning: ‘KimeraRPGO::RobustSolverParams::incremental’ will be initialized after [-Wreorder]
   bool incremental;
        ^~~~~~~~~~~
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/include/KimeraRPGO/SolverParams.h:121:8: warning:   ‘bool KimeraRPGO::RobustSolverParams::log_output’ [-Wreorder]
   bool log_output;
        ^~~~~~~~~~
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/include/KimeraRPGO/SolverParams.h:28:3: warning:   when initialized here [-Wreorder]
   RobustSolverParams()
   ^~~~~~~~~~~~~~~~~~
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp: In member function ‘bool KimeraRPGO::GenericSolver::addAndCheckIfOptimize(const gtsam::NonlinearFactorGraph&, const gtsam::Values&)’:
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:42:8: warning: unused variable ‘do_optimize’ [-Wunused-variable]
   bool do_optimize = true;
        ^~~~~~~~~~~
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp: In member function ‘void KimeraRPGO::GenericSolver::removePriorsWithPrefix(const char&)’:
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:114:44: error: ‘PriorFactor’ is not a member of ‘gtsam’
     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
                                            ^~~~~~~~~~~
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:114:44: note: suggested alternative: ‘Factor’
     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
                                            ^~~~~~~~~~~
                                            Factor
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:114:44: error: ‘PriorFactor’ is not a member of ‘gtsam’
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:114:44: note: suggested alternative: ‘Factor’
     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
                                            ^~~~~~~~~~~
                                            Factor
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:114:16: error: parse error in template argument list
     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:114:68: error: no match for ‘operator>’ (operand types are ‘<unresolved overloaded function type>’ and ‘boost::shared_ptr<gtsam::NonlinearFactor>’)
     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from /usr/include/boost/lexical_cast/detail/converter_lexical.hpp:50:0,
                 from /usr/include/boost/lexical_cast/try_lexical_convert.hpp:42,
                 from /usr/include/boost/lexical_cast.hpp:32,
                 from /usr/include/boost/date_time/string_parse_tree.hpp:13,
                 from /usr/include/boost/date_time/period_parser.hpp:14,
                 from /usr/include/boost/date_time/date_facet.hpp:23,
                 from /usr/include/boost/date_time/gregorian/gregorian_io.hpp:16,
                 from /usr/include/boost/date_time/gregorian/gregorian.hpp:31,
                 from /usr/include/boost/date_time/posix_time/time_formatters.hpp:12,
                 from /usr/include/boost/date_time/posix_time/posix_time.hpp:24,
                 from /home/ibrahimkaramavus/catkin_ws/devel/include/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:26,
                 from /home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:16:
/usr/include/boost/array.hpp:336:10: note: candidate: template<class T, long unsigned int N> bool boost::operator>(const boost::array<T, sz>&, const boost::array<T, sz>&)
     bool operator> (const array<T,N>& x, const array<T,N>& y) {
          ^~~~~~~~
/usr/include/boost/array.hpp:336:10: note:   template argument deduction/substitution failed:
/home/ibrahimkaramavus/catkin_ws/src/Kimera-RPGO/src/GenericSolver.cpp:114:77: note:   ‘boost::shared_ptr<gtsam::NonlinearFactor>’ is not derived from ‘const boost::array<T, sz>’
     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
                                                                             ^
In file included from /usr/include/boost/variant/variant.hpp:42:0,
[... 384 lines cut ...]
.....................................................................................................................................................................................
Failed     << kimera_rpgo:make                                  [ Exited with code 2 ]                                                                                               
Failed    <<< kimera_rpgo                                       [ 50.4 seconds ]                                                                                                     
Abandoned <<< kimera_vio                                        [ Unrelated job failed ]                                                                                             
Abandoned <<< kimera_vio_ros                                    [ Unrelated job failed ]                                                                                             
Abandoned <<< kimera_semantics_ros                              [ Unrelated job failed ]                                                                                             
Finished  <<< voxblox_ros                                       [ 51.8 seconds ]                                                                                                     
[build] Summary: 44 of 48 packages succeeded.                                                                                                                                        
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                                                                     
[build]   Warnings:  2 packages succeeded with warnings.                                                                                                                             
[build]   Abandoned: 3 packages were abandoned.                                                                                                                                      
[build]   Failed:    1 packages failed.                    

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): No

yunzc, 2021-01-09:
@IbrahimKaramavus1773 pull the latest master from KimeraRPGO, [this commit](https://github.com/MIT-SPARK/Kimera-RPGO/commit/3003ed918f11c3f45ccc98e8726839c5f6fc8b9f) should hopefully solve the problem.

pooja-kabra, 2023-05-17:
I am working with Ubuntu 20.04 and ROS Foxy. 
Successfuly built GTSAM on develop branch with BUILD_WITH_MARCH_NATIVE = OFF and installed it.

Used out of the box cmake options for Kimera-RPGO, see image:
[image]

```
pooja@pooja-ThinkPad-P1-Gen-5:~/Kimera-RPGO/build$ make install
[  6%] Building CXX object CMakeFiles/KimeraRPGO.dir/src/GenericSolver.cpp.o
/home/pooja/Kimera-RPGO/src/GenericSolver.cpp: In member function ‘void KimeraRPGO::GenericSolver::removePriorsWithPrefix(const char&)’:
/home/pooja/Kimera-RPGO/src/GenericSolver.cpp:138:77: error: no matching function for call to ‘dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3> >(std::shared_ptr<gtsam::NonlinearFactor>&)’
  138 |     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
      |                                                                             ^
In file included from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/format/alt_sstream.hpp:22,
                 from /usr/include/boost/format/internals.hpp:24,
                 from /usr/include/boost/format.hpp:38,
                 from /home/pooja/Kimera-RPGO/include/KimeraRPGO/Logger.h:8,
                 from /home/pooja/Kimera-RPGO/include/KimeraRPGO/GenericSolver.h:16,
                 from /home/pooja/Kimera-RPGO/src/GenericSolver.cpp:9:
/usr/include/boost/smart_ptr/shared_ptr.hpp:963:42: note: candidate: ‘template<class T, class U> boost::shared_ptr<X> boost::dynamic_pointer_cast(const boost::shared_ptr<U>&)’
  963 | template<class T, class U> shared_ptr<T> dynamic_pointer_cast( shared_ptr<U> const & r ) BOOST_SP_NOEXCEPT
      |                                          ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/smart_ptr/shared_ptr.hpp:963:42: note:   template argument deduction/substitution failed:
/home/pooja/Kimera-RPGO/src/GenericSolver.cpp:138:77: note:   ‘std::shared_ptr<gtsam::NonlinearFactor>’ is not derived from ‘const boost::shared_ptr<U>’
  138 |     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
      |                                                                             ^
In file included from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/format/alt_sstream.hpp:22,
                 from /usr/include/boost/format/internals.hpp:24,
                 from /usr/include/boost/format.hpp:38,
                 from /home/pooja/Kimera-RPGO/include/KimeraRPGO/Logger.h:8,
                 from /home/pooja/Kimera-RPGO/include/KimeraRPGO/GenericSolver.h:16,
                 from /home/pooja/Kimera-RPGO/src/GenericSolver.cpp:9:
/usr/include/boost/smart_ptr/shared_ptr.hpp:1005:42: note: candidate: ‘template<class T, class U> boost::shared_ptr<X> boost::dynamic_pointer_cast(boost::shared_ptr<U>&&)’
 1005 | template<class T, class U> shared_ptr<T> dynamic_pointer_cast( shared_ptr<U> && r ) BOOST_SP_NOEXCEPT
      |                                          ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/smart_ptr/shared_ptr.hpp:1005:42: note:   template argument deduction/substitution failed:
/home/pooja/Kimera-RPGO/src/GenericSolver.cpp:138:77: note:   ‘std::shared_ptr<gtsam::NonlinearFactor>’ is not derived from ‘boost::shared_ptr<U>’
  138 |     if (boost::dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3>>(factor)) {
      |                                                                             ^
/home/pooja/Kimera-RPGO/src/GenericSolver.cpp:141:21: error: no matching function for call to ‘dynamic_pointer_cast<gtsam::PriorFactor<gtsam::Pose3> >(std::shared_ptr<gtsam::NonlinearFactor>&)’
  141 |               factor);
      |                     ^
In file included from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/format/alt_sstream.hpp:22,
                 from /usr/include/boost/format/internals.hpp:24,
                 from /usr/include/boost/format.hpp:38,
                 from /home/pooja/Kimera-RPGO/include/KimeraRPGO/Logger.h:8,
                 from /home/pooja/Kimera-RPGO/include/KimeraRPGO/GenericSolver.h:16,
                 from /home/pooja/Kimera-RPGO/src/GenericSolver.cpp:9:
/usr/include/boost/smart_ptr/shared_ptr.hpp:963:42: note: candidate: ‘template<class T, class U> boost::shared_ptr<X> boost::dynamic_pointer_cast(const boost::shared_ptr<U>&)’
  963 | template<class T, class U> shared_ptr<T> dynamic_pointer_cast( shared_ptr<U> const & r ) BOOST_SP_NOEXCEPT
      |                                          ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/smart_ptr/shared_ptr.hpp:963:42: note:   template argument deduction/substitution failed:
/home/pooja/Kimera-RPGO/src/GenericSolver.cpp:141:21: note:   ‘std::shared_ptr<gtsam::NonlinearFactor>’ is not derived from ‘const boost::shared_ptr<U>’
  141 |               factor);
      |                     ^
In file included from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/format/alt_sstream.hpp:22,
                 from /usr/include/boost/format/internals.hpp:24,
                 from /usr/include/boost/format.hpp:38,
                 from /home/pooja/Kimera-RPGO/include/KimeraRPGO/Logger.h:8,
                 from /home/pooja/Kimera-RPGO/include/KimeraRPGO/GenericSolver.h:16,
                 from /home/pooja/Kimera-RPGO/src/GenericSolver.cpp:9:
/usr/include/boost/smart_ptr/shared_ptr.hpp:1005:42: note: candidate: ‘template<class T, class U> boost::shared_ptr<X> boost::dynamic_pointer_cast(boost::shared_ptr<U>&&)’
[... cut ...]
  148 |               factor);
      |                     ^
make[2]: *** [CMakeFiles/KimeraRPGO.dir/build.make:76: CMakeFiles/KimeraRPGO.dir/src/GenericSolver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:212: CMakeFiles/KimeraRPGO.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```
The change in https://github.com/MIT-SPARK/Kimera-RPGO/commit/3003ed918f11c3f45ccc98e8726839c5f6fc8b9f mentioned by @yunzc is already present on the develop branch, but build still fails.

Any help would be appreciated thanks!