MIT-SPARK/Kimera-VIO-ROS #125: failed to "Initializing pose graph visualizer"

JayShaw-git, 2021-01-24:
**Description:**
 run the example of online with Euroc datasets

**Command:**
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch
```

**Console output:**
```
... logging to /home/jayshaw/.ros/log/c430d344-5e34-11eb-9b41-a4bb6d571a89/roslaunch-jayshaw-20046.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jayshaw:33509/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/jayshaw/cat...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [20061]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [20062]
[ INFO] [1611487885.829152823]: Initializing pose graph visualizer
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 20061, exit code -11, cmd /home/jayshaw/catkin_kvio/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/jayshaw/catkin_kvio/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/jayshaw/catkin_kvio/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/jayshaw/catkin_kvio/src/Kimera-VIO/params/Euroc/flags/VioBackEnd.flags --flagfile=/home/jayshaw/catkin_kvio/src/Kimera-VIO/params/Euroc/flags/RegularVioBackEnd.flags --flagfile=/home/jayshaw/catkin_kvio/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/jayshaw/catkin_kvio/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true left_cam/image_raw:=/cam0/image_raw right_cam/image_raw:=/cam1/image_raw imu:=/imu0 left_cam/camera_info:=/cam0/camera_info right_cam/camera_info:=/cam1/camera_info reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/jayshaw/.ros/log/c430d344-5e34-11eb-9b41-a4bb6d571a89/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/jayshaw/.ros/log/c430d344-5e34-11eb-9b41-a4bb6d571a89/kimera_vio_ros-kimera_vio_ros_node-1*.log

```
Hope you can help me ,thank you very much!
**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used :V5.0
* GTSAM version used: 4.00
* OpenGV version used:
* OpenCV version used: type `opencv_version`3.4.2
* Operating system and version : Ubuntu 18.04 
* Did you change the source code? (yes / no): no
*

shaheer34mts, 2021-03-27:
Hey, did you find a solution?

moksh019, 2021-10-20:
sorry for late reply, you can refer this #162