MIT-SPARK/Kimera-VIO-ROS #126: "Missing landmark information to build 3D Mesh" using Realsense D435i

mtmercado, 2021-02-03:
Hello! Thank you for your work on this library! It is very interesting! I started playing around with it and in need of help. I'm trying use Kimera VIO and Semantics with Intel Realsense D435i. However, I'm encountering this warning whenever I run the kimera_vio launch file:

`W0203 14:13:26.420764  4725 Mesher.cpp:1380] Missing landmark information to build 3D Mesh.`

I tried to run kimera_semantics as well however, I'm not getting the desired result. I'm assuming the 3D mesh is a data required in Kimera Semantics, is that right? If so, how can I specify the landmark information for the 3D mesh to build successfully?

Thank you!

Logs:
_kimera vio_
```
ailab-cv@ailabcv-NUC8i7BEH:~$ roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch run_stereo_dense:=true
... logging to /home/ailab-cv/.ros/log/6d4b41b6-65de-11eb-a604-04ea56db17cb/roslaunch-ailabcv-NUC8i7BEH-4689.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ailabcv-NUC8i7BEH:39771/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: left_cam
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/ailab-cv/ca...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: right_cam
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: False

NODES
  /
    velo_link_broadcaster (tf/static_transform_publisher)
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [4704]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [4705]
process[velo_link_broadcaster-3]: started with pid [4706]
[ INFO] [1612329205.947763145]: Initializing pose graph visualizer
I0203 14:13:26.096387  4704 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0203 14:13:26.096485  4704 PipelineModule.h:449] MISO Pipeline Module: Display has no output queue registered.
I0203 14:13:26.096644  4704 Pipeline.cpp:479] Pipeline Modules launched (parallel_run set to 1).
I0203 14:13:26.096861  4729 Pipeline.h:199] Spinning Kimera-VIO.
I0203 14:13:26.096849  4704 KimeraVioRos.cpp:117] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Display [ms]                            	    0	
VioBackEnd [ms]                         	    0	
VioFrontEnd [ms]                        	    0	
Visualizer [ms]                         	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	    0	
data_provider_right_frame_queue Size [#]	    0	
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
[... 102 lines cut ...]

NODES
  /
    cloudify (nodelet/nodelet)
    kimera_semantics_node (voxblox_ros/tsdf_server)
    nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[nodelet_manager-1]: started with pid [5677]
process[cloudify-2]: started with pid [5678]
process[kimera_semantics_node-3]: started with pid [5679]
[ INFO] [1612330048.302067468]: Loading nodelet /cloudify of type depth_image_proc/point_cloud_xyzrgb to manager nodelet_manager with the following remappings:
[ INFO] [1612330048.302637143]: /depth_registered/image_rect -> /kimera/depth_registered/image_rect_raw
[ INFO] [1612330048.302654935]: /depth_registered/points -> /semantic_pointcloud
[ INFO] [1612330048.302670181]: /rgb/camera_info -> /camera/color/camera_info
[ INFO] [1612330048.302684132]: /rgb/image_rect_color -> /camera/color/image_raw
[ INFO] [1612330048.303633830]: waitForService: Service [/nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1612330048.322386107]: Initializing nodelet with 8 worker threads.
[ INFO] [1612330048.324319301]: waitForService: Service [/nodelet_manager/load_nodelet] is now available.
[ INFO] [1612330048.497208818]: Updating mesh.
[ INFO] [1612330048.597140305]: Updating mesh.
[ INFO] [1612330048.697146803]: Updating mesh.
[ INFO] [1612330048.797159197]: Updating mesh.
```

Ravi3191, 2021-03-25:
I am having the same problem here. Did you resolve the issue?

BBlumi97, 2021-11-10:
I am having the same issue. Did anyone found a solution for this ?