MIT-SPARK/Kimera-VIO-ROS #128: Error with Kimera-VIO ROS wrapper

mohammad-h-b, 2021-02-22:
**Description:**
I've installed Kimera (semantics & vio-ROS) and also have an Intel Realsense D435i and followed the steps of getting kimera_semantics working in real-time using D435i camera; however, for some reason I don't get the semantic map. 

**Command:**
```
roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch
```

**Console output:**
```
mohammad@mohammad:~/catkin_ws$ roslaunch kimera_vio_ros kimera_vio_ros_alsense_IR.launch
... logging to /home/mohammad/.ros/log/1719bfc8-74b8-11eb-8f79-c8e2658ec50b/roslaunch-mohammad-6711.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mohammad:32781/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: left_cam
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/mohammad/ca...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: right_cam
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: False

NODES
  /
    velo_link_broadcaster (tf/static_transform_publisher)
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [6726]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [6727]
process[velo_link_broadcaster-3]: started with pid [6728]
[ INFO] [1613961960.559582842]: Initializing pose graph visualizer
I0221 21:46:00.687762  6726 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0221 21:46:00.687829  6726 PipelineModule.h:449] MISO Pipeline Module: Display has no output queue registered.
I0221 21:46:00.687911  6726 Pipeline.cpp:479] Pipeline Modules launched (parallel_run set to 1).
I0221 21:46:00.688042  6726 KimeraVioRos.cpp:117] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Mesher [ms]                             	    0	
VioBackEnd [ms]                         	    0	
VioFrontEnd [ms]                        	    0	
Visualizer [ms]                         	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	    0	
data_provider_right_frame_queue Size [#]	    0	
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
visualizer_frontend_queue Size [#]      	    0	
visualizer_mesher_queue Size [#]        	    0	
I0221 21:46:00.688122  6751 Pipeline.h:199] Spinning Kimera-VIO.
I0221 21:46:01.688239  6726 KimeraVioRos.cpp:117] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Data Provider [ms]                      	    0	
Display [ms]                            	    0	
Mesher [ms]                             	    0	
VioBackEnd [ms]                         	    0	
VioFrontEnd [ms]                        	    0	
Visualizer [ms]                         	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	    0	
data_provider_right_frame_queue Size [#]	    0	
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
visualizer_frontend_queue Size [#]      	    0	
visualizer_mesher_queue Size [#]        	    0	
I0221 21:46:02.688254  6726 KimeraVioRos.cpp:117] Statistics
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

ToniRV, 2021-02-23:
Are you using a rosbag?
If you are, then make sure you set the clock flag: `rosbag play ... --clock` 
If you aren't, then make sure you disable sim_time in the launch file of kimera_vio_ros_realsense_IR.launch.
Just search for `use_sim_time` flag and set it to `false`.

mohammad-h-b, 2021-02-23:
I'm not using rosbag. the `use_sim_time` is set to false by default in the launch file:
```
<!-- Set use_sim_time to true if you use rosbag with clock argument -->
  <arg name="should_use_sim_time" default="false"/>
```