MIT-SPARK/Kimera-VIO-ROS #132: Catkin build error

Viky397, 2021-02-23:
Hello, when I `catkin build` Kimera-VIO-ROS, I get the following error. Any help is appreciated, thank you!

[image]

Viky397, 2021-02-23:
I see: https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/115
I am working in ROS Noetic and Ubuntu 20.04

neo-knight-td, 2021-03-09:
Same problem here, did you find a solution ?

YukiTbst, 2021-03-11:
me too, also ROS Noetic and Ubuntu 20.04. maybe it doesn't work on ROS Noetic

geothan, 2021-09-14:
Same problem here, did you find any solution ?

iampaulidrobo, 2021-10-08:
Does it work now?
I am trying on ros noetic and there is opencv error.Although I followed the instruction from:
https://github.com/MIT-SPARK/Kimera-VIO/blob/master/docs/kimera_vio_install.md

Shrioe, 2024-06-17:
Hello, I'm facing teh same build error too, may I ask did you mange to find a solution for that? Can you give me skme suggestions? Thank you!