MIT-SPARK/Kimera-VIO-ROS #133: Kimera VIO ROS Wrapper Crashes after ~20 Seconds

njcich, 2021-02-27:
Hello, I'm working on a project with an Intel D435i camera, and am attempting to run the Kimera ROS wrapper alongside the D435i ROS wrapper on an NVIDIA Jetson AGX Xavier. The only two ROS packages that I've launched are the D435i wrapper (which I haven't had issues with) and the Kimera wrapper. The Kimera wrapper works for about twenty seconds--I've been able to print out odometry measurements that seem correct--but then suddenly stops publishing any information. The error message describing the cause of this seems to be "Backend did not return an output: shutting down backend." I'm unsure of how to proceed. Any help is appreciated!

Note: the Kimera wrapper continues to run after the error message below (final line of the output that I have attached), but no longer publishes any information.

`roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch
... logging to /home/unitree/.ros/log/4eb76b04-7889-11eb-84db-48b02d0790cc/roslaunch-nx-31707.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:35337/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: left_cam
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/unitree/cat...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: right_cam
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: False

NODES
  /
    velo_link_broadcaster (tf/static_transform_publisher)
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [31723]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [31724]
[ INFO] [1614384171.746155098]: Initializing pose graph visualizer
process[velo_link_broadcaster-3]: started with pid [31730]
I0226 19:02:52.765061 31723 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0226 19:02:52.765475 31723 PipelineModule.h:449] MISO Pipeline Module: Display has no output queue registered.
I0226 19:02:52.766374 31723 Pipeline.cpp:479] Pipeline Modules launched (parallel_run set to 1).
I0226 19:02:52.768501 31723 KimeraVioRos.cpp:117] Statistics
-----------                                  # Log Hz {avg     +- std    } [min,max]
Display [ms]                                0
Mesher [ms]                                0
VioBackEnd [ms]                            0
VioFrontEnd [ms]                            0
Visualizer [ms]                            0
backend_input_queue Size [#]                0
data_provider_left_frame_queue Size [#]    0
data_provider_right_frame_queue Size [#]    0
display_input_queue Size [#]                0
mesher_backend Size [#]                    0
mesher_frontend Size [#]                    0
stereo_frontend_input_queue Size [#]        0
visualizer_backend_queue Size [#]          0
visualizer_frontend_queue Size [#]          0
visualizer_mesher_queue Size [#]            0
I0226 19:02:52.768951 31756 Pipeline.h:199] Spinning Kimera-VIO.
I0226 19:02:53.403529 31748 VioBackEnd.h:167] ------------------- Initialize Pipeline: timestamp = 1614384173096156597--------------------
W0226 19:02:53.403997 31748 InitializationFromImu.cpp:25] InitializationFromImu: assumes that the vehicle is stationary and upright along some axis,and gravity vector is along a single axis!
[... 894 lines cut ...]
Slot # 290: Smart Factor (valid: NO!, deg: no isCheir: no): Factor  x49 x50 x51 x52 x53
Slot # 291: Smart Factor (valid: NO!, deg: YES! isCheir: no): Factor  x49 x50 x51 x52 x53
Slot # 292: Smart Factor (valid: NO!, deg: YES! isCheir: no): Factor  x31 x32 x33 x34 x35 x36 x37 x38 x39 x40 x41 x42 x43 x44 x45 x46 x47 x48 x49 x50 x51 x52 x53
Slot # 293: Smart Factor (valid: NO!, deg: YES! isCheir: no): Factor  x49 x50 x51 x52 x53
Slot # 294: Smart Factor (valid: NO!, deg: YES! isCheir: no): Factor  x49 x50 x51 x52 x53
Slot # 295: Smart Factor (valid: NO!, deg: YES! isCheir: no): Factor  x31 x32 x33 x34 x35 x36 x37 x38 x39 x40 x41 x42 x43 x44 x45 x46 x47 x48 x49 x50 x51 x52 x53
Slot # 296: Smart Factor (valid: NO!, deg: YES! isCheir: no): Factor  x49 x50 x51 x52 x53
Slot # 297: Smart Factor (valid: NO!, deg: YES! isCheir: no): Factor  x4 x5 x6 x7 x8 x9 x10 x11 x12 x13 x14 x15 x16 x17 x18 x19 x20 x21 x22 x23 x24 x25 x26 x27 x28 x29 x30 x31 x32 x33 x34 x35 x36 x37 x38 x39 x40 x41 x42 x43 x44 x45 x46 x47 x48 x49 x50 x51 x52 x53
Slot # 298: Smart Factor (valid: NO!, deg: YES! isCheir: no): Factor  x49 x50 x51 x52 x53

I0226 19:03:02.174692 31748 VioBackEnd.cpp:1709]  ]
I0226 19:03:02.174732 31748 VioBackEnd.cpp:1712] Nr deleted slots: 0, with slots:
I0226 19:03:02.174757 31748 VioBackEnd.cpp:1714] [

I0226 19:03:02.174788 31748 VioBackEnd.cpp:1741]  ]
I0226 19:03:02.174808 31748 VioBackEnd.cpp:1744] Nr of values in state_ : 93, with keys:
[
b22 b23 b24 b25 b26 b27 b28 b29 b30 b31 b32 b33 b34 b35 b36 b37 b38 b39 b40 b41 b42 b43 b44 b45 b46 b47 b48 b49 b50 b51 b52 v22 v23 v24 v25 v26 v27 v28 v29 v30 v31 v32 v33 v34 v35 v36 v37 v38 v39 v40 v41 v42 v43 v44 v45 v46 v47 v48 v49 v50 v51 v52 x22 x23 x24 x25 x26 x27 x28 x29 x30 x31 x32 x33 x34 x35 x36 x37 x38 x39 x40 x41 x42 x43 x44 x45 x46 x47 x48 x49 x50 x51 x52
I0226 19:03:02.175400 31748 VioBackEnd.cpp:1750]  ]
I0226 19:03:02.175421 31748 VioBackEnd.cpp:1753] Nr values in new_values_ : 3, with keys:
[
b53  v53  x53
I0226 19:03:02.175482 31748 VioBackEnd.cpp:1761]  ]
I0226 19:03:02.175499 31748 VioBackEnd.cpp:1772]  =============== END: CATCHING EXCEPTION ===============
E0226 19:03:02.176074 31748 VioBackEndModule.cpp:34] Backend did not return an output: shutting down backend.`

kkoledic1212, 2021-04-27:
Did you perhaps find the solution to this problem?

njcich, 2021-04-28:
I am running Kimera on a quadrupedal robot, and by starting the robot's movement before launching Kimera, I am usually able to avoid this issue. It doesn't work every time, but it usually will stop Kimera from crashing.

martinakos, 2021-07-07:
I also have this exception. Is there any way to avoid it starting from a stationary camera?

soldierofhell, 2021-09-13:
Also got this error. I noticed that on my rosbag it appears when robot starts to move. I thought that starting from stationary state is the preferred way to use VIO to properly initialize imu, see for example this message at the beginning
```
W0913 10:19:27.229720 28225 InitializationFromImu.cpp:25] InitializationFromImu: assumes that the vehicle is stationary and upright along some axis,and gravity vector is along a single axis!
```
This error seems related to gtsam. Also first notification is from 2021, so I wonder if maybe downgrade of gtsam could help, because rosinstall clone "develop" branch. Anybody tried?

Regards,

soldierofhell, 2021-09-13:
The error is triggered by this [smoother_->update()](https://github.com/MIT-SPARK/Kimera-VIO/blob/641576fd86bdecbd663b4db3cb068f49502f3a2c/src/backend/VioBackend.cpp#L1212-L1213) and originates in gtsam [Values.cpp](https://github.com/borglab/gtsam/blob/cd3854a1f6db923d40ecf3ced56bafbe339d1b3c/gtsam/nonlinear/Values.cpp#L224-L230)

soldierofhell, 2021-09-13:
Here is an [issue](https://github.com/borglab/gtsam/issues/301) on gtsam repo also with smart factors, but might not be related

BBlumi97, 2021-11-08:
Hi all,

Did anyone found a solution for this error? 
I am facing the same error that my kimera crashes when I start moving my camera.

Regards,
Benjamin