MIT-SPARK/Kimera-VIO-ROS #136: roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch problem

wuyang171, 2021-04-19:
**Description:**

when i compile the project,using ros to run.probem accured.
**Command:**
```
# replace this line with the command(s) you used
```

**Console output:**
```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://y***-OMEN-Laptop-15-ek0xxx:39623/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/yang/catkin...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: 
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [24832]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [24833]
[ INFO] [1618824283.152697517]: Initializing pose graph visualizer
I0419 17:24:43.274405 24832 Logger.cpp:59] Opening output file: traj_gt.csv
I0419 17:24:43.275686 24832 RosBagDataProvider.cpp:52] Constructing RosbagDataProvider from path: 
 - Rosbag Path: 
With ROS topics: 
 - Left cam: /cam0/image_raw
 - Right cam: /cam1/image_raw
 - IMU: /imu0
 - GT odom: 
F0419 17:24:43.275704 24832 RosBagDataProvider.cpp:60] Check failed: !rosbag_path_.empty() 
*** Check failure stack trace: ***
    @     0x7f60a990fb0d  google::LogMessage::Fail()
    @     0x7f60a99119b1  google::LogMessage::SendToLog()
    @     0x7f60a990f63d  google::LogMessage::Flush()
    @     0x7f60a9912389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f60a90d7158  VIO::RosbagDataProvider::RosbagDataProvider()
    @     0x7f60a90c3a23  VIO::KimeraVioRos::createDataProvider()
    @     0x7f60a90c457f  VIO::KimeraVioRos::runKimeraVio()
    @     0x56198b53ab2a  (unknown)
    @     0x7f60a7b12bf7  __libc_start_main
    @     0x56198b53b03a  (unknown)
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 24832, exit code -6, cmd /home/yang/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/yang/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/yang/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/yang/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/yang/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/yang/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/yang/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/yang/.ros/log/2c9427f0-a0ee-11eb-b0b1-2cdb07139caf/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/yang/.ros/log/2c9427f0-a0ee-11eb-b0b1-2cdb07139caf/kimera_vio_ros-kimera_vio_ros_node-1*.log

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

ToniRV, 2021-04-21:
You need to provide the filepath to the rosbag you want to use:
```
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: 
 ```
 Just modify the parameter `rosbag_path` in the launch file you are using.

closed 2021-04-21 by ToniRV

Abirami-Segar, 2021-06-04:
hi

could you please explain where i can find this * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: ?

I looked for the RosbagDataProvider.cpp program, but could not edit the path of the rosbag. I am new to ROS. any help would be appreciated. Thanks in advance!

LimHyungTae, 2024-08-17:
## Error

After updating `rosbag_path`, the following error might happen:

```
F0817 15:37:33.890363 3954893 RosBagDataProvider.cpp:304] Check failed: vio_params_.backend_params_->autoInitialize_ == 1 (0 vs. 1) Requested ground-truth initialization, but no gt_odom_topic was given. Make sure you set ground_truth_odometry_rosbag_topic, or turn autoInitialize to false in BackendParameters.yaml.
*** Check failure stack trace: ***
    @     0x7fad5455f703  google::LogMessage::Fail()
    @     0x7fad545644ab  google::LogMessage::SendToLog()
    @     0x7fad5455f3ff  google::LogMessage::Flush()
    @     0x7fad5455fc2f  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fad542fe215  VIO::RosbagDataProvider::parseRosbag()
    @     0x7fad542fe685  VIO::RosbagDataProvider::initialize()
    @     0x7fad542e5a04  VIO::KimeraVioRos::createDataProvider()
    @     0x7fad542e76ed  VIO::KimeraVioRos::runKimeraVio()
    @     0x55f4cc0539c9  main
    @     0x7fad53a35083  __libc_start_main
    @     0x55f4cc053a4e  _start
================================================================================REQUIRED process [kimera_vio_ros/kimera_vio_ros_node-1] has died!
process has died [pid 3954893, exit code -6, cmd /home/shapelim/kimera_vio_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/shapelim/kimera_vio_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/shapelim/kimera_vio_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/shapelim/kimera_vio_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/shapelim/kimera_vio_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/shapelim/kimera_vio_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --use_external_odometry=false --do_coarse_imu_camera_temporal_sync=false --do_fine_imu_camera_temporal_sync=false --lcd_no_optimize=false --lcd_no_detection=false --depth_image_mask= --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --minloglevel=0 --v=1 --log_output=false --log_euroc_gt_data=false --output_path=/home/shapelim/kimera_vio_ws/src/Kimera-VIO-ROS/output_logs/ --lcd_disable_stereo_match_depth_check=false --no_incremental_pose=false --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/shapelim/.ros/log/e8f4c314-5ccf-11ef-8b10-89c894c2eed9/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/shapelim/.ros/log/e8f4c314-5ccf-11ef-8b10-89c894c2eed9/kimera_vio_ros-kimera_vio_ros_node-1*.log
Initiating shutdown!
================================================================================
```
[image]

## Solution

Change `autoInitialize: 0` to `autoInitialize: 1` in the `params/Euroc/BackendParams.yaml` file.