MIT-SPARK/Kimera-VIO-ROS #137: Catkin build error build/kimera_vio/testKimeraVIO: symbol lookup error

Goldsaints, 2021-04-19:
Thank you for your excellent work and I am very interested in it. I download your code and install it, but an error encountered.

**Command:**
```
ubuntu:~/catkin_vio/src$ catkin build
```

**Console output:**
```
/home/ben/catkin_vio/build/kimera_vio/testKimeraVIO: symbol lookup error: /home/ben/catkin_ws/devel/lib/libkimera_vio.so: undefined symbol: _ZNK5gtsam16NoiseModelFactor5printERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKN5boost8functionIFS6_mEEE
CMake Error at /usr/share/cmake-3.10/Modules/GoogleTestAddTests.cmake:39 (message):
  Error running test executable.

    Path: '/home/ben/catkin_vio/build/kimera_vio/testKimeraVIO'
    Result: 127
    Output:
      

make[2]: *** [testKimeraVIO] Error 1
make[2]: *** Deleting file“testKimeraVIO”
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
make: *** [all] Error 2
```
I get the dependencies of the project by
```
wstool merge Kimera-VIO-ROS/install/kimera_vio_ros_https.rosinstall
```
Could you help me solve it? Sincerely.

Tumucin, 2021-04-19:
Hello

I have the same problem. Did you manage to solve system lookup error?

Goldsaints, 2021-04-19:
> Hello
> 
> I have the same problem. Did you manage to solve system lookup error?

Sorry, not yet.

mikwaluk, 2021-04-20:
Hi, I experienced the same problem, opened #138

mikwaluk, 2021-04-21:
Figured out a solution, although not exactly sure if it's correct this way. But at least it works and Kimera produces the correct estimation :)

```
diff --git a/src/KimeraVioRos.cpp b/src/KimeraVioRos.cpp
index 68fb81d..dd6343c 100644
--- a/src/KimeraVioRos.cpp
+++ b/src/KimeraVioRos.cpp
@@ -121,15 +121,15 @@ bool KimeraVioRos::spin() {
     std::future<bool> data_provider_handle =
         std::async(std::launch::async,
                    &VIO::RosDataProviderInterface::spin,
-                   std::ref(*CHECK_NOTNULL(data_provider_.get())));
+                   CHECK_NOTNULL(data_provider_.get()));
     std::future<bool> vio_viz_handle =
         std::async(std::launch::async,
                    &VIO::Pipeline::spinViz,
-                   std::ref(*CHECK_NOTNULL(vio_pipeline_.get())));
+                   CHECK_NOTNULL(vio_pipeline_.get()));
     std::future<bool> vio_pipeline_handle =
         std::async(std::launch::async,
                    &VIO::Pipeline::spin,
-                   std::ref(*CHECK_NOTNULL(vio_pipeline_.get())));
+                   CHECK_NOTNULL(vio_pipeline_.get()));
```

ToniRV, 2021-04-21:
@mikwaluk that should be the way to do it, feel free to open a PR and I'll merge asap!

closed 2021-04-21 by ToniRV

Abirami-Segar, 2021-06-03:
Hi 

I faced the same problem as you did while compiling the packages. I was not able to build Kimera_VIO_ROS. I added the suggestion you gave.. Even then, there seems to be error while building this particular package. Could you please help?