MIT-SPARK/Kimera-VIO-ROS #141: roslaunch error

huziqi, 2021-04-24:
**Description:**
Ubuntu 18.04
ROS: melodic
When i tried to run roslaunch kimera_vio_ros_euroc.launch, i got such output errors:

**Command:**
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch
```

**Console output:**
```

... logging to /home/hzq/.ros/log/d81405e8-a4f7-11eb-b60f-c46516114b0c/roslaunch-hzq-19301.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hzq:45941/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/hzq/my_spac...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: /home/hzq/dataset...
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

auto-starting new master
process[master]: started with pid [19311]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d81405e8-a4f7-11eb-b60f-c46516114b0c
process[rosout-1]: started with pid [19322]
started core service [/rosout]
process[kimera_vio_ros/kimera_vio_ros_node-2]: started with pid [19329]
process[kimera_vio_ros/posegraph_viewer-3]: started with pid [19330]
[ INFO] [1619266996.333315824]: Initializing pose graph visualizer
I0424 20:23:16.524582 19329 Logger.cpp:59] Opening output file: traj_gt.csv
I0424 20:23:16.525610 19329 RosBagDataProvider.cpp:52] Constructing RosbagDataProvider from path: 
 - Rosbag Path: /home/hzq/dataset/euroc/MH_01_easy.bag
With ROS topics: 
 - Left cam: /cam0/image_raw
 - Right cam: /cam1/image_raw
 - IMU: /imu0
 - GT odom: 
I0424 20:23:16.526614 19329 RosBagDataProvider.cpp:83] Initialize Rosbag Data Provider.
I0424 20:23:16.526623 19329 RosBagDataProvider.cpp:201] Parsing rosbag data.
F0424 20:23:16.547317 19329 RosBagDataProvider.cpp:222] Check failed: vio_params_.backend_params_->autoInitialize_ == 1 (0 vs. 1) Requested ground-truth initialization, but no gt_odom_topic was given. Make sure you set ground_truth_odometry_rosbag_topic, or turn autoInitialize to false in BackendParameters.yaml.
*** Check failure stack trace: ***
    @     0x7f01cd846b0d  google::LogMessage::Fail()
    @     0x7f01cd8489b1  google::LogMessage::SendToLog()
    @     0x7f01cd84663d  google::LogMessage::Flush()
    @     0x7f01cd849389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f01cd00c8e8  VIO::RosbagDataProvider::parseRosbag()
    @     0x7f01cd00d121  VIO::RosbagDataProvider::initialize()
    @     0x7f01ccffaa2b  VIO::KimeraVioRos::createDataProvider()
    @     0x7f01ccffb57f  VIO::KimeraVioRos::runKimeraVio()
    @     0x561b23a77b2a  (unknown)
    @     0x7f01cbca4bf7  __libc_start_main
    @     0x561b23a7803a  (unknown)
[kimera_vio_ros/kimera_vio_ros_node-2] process has died [pid 19329, exit code -6, cmd /home/hzq/my_space/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/hzq/my_space/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/hzq/my_space/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/hzq/my_space/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/hzq/my_space/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/hzq/my_space/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/hzq/my_space/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/hzq/.ros/log/d81405e8-a4f7-11eb-b60f-c46516114b0c/kimera_vio_ros-kimera_vio_ros_node-2.log].
log file: /home/hzq/.ros/log/d81405e8-a4f7-11eb-b60f-c46516114b0c/kimera_vio_ros-kimera_vio_ros_node-2*.log

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used:lastest master
* GTSAM version used:default
* OpenGV version used:default
* OpenCV version used: 3.4.1
* Operating system and version Ubuntu 18.04
* Did you change the source code? (yes / no): no

huziqi, 2021-04-24:
when i look into source code, i found the var 'autoInitialize' has been set to 0. why the roslaunch still wants me to set it to false in BackendParameters.yaml?

zheng-xiao, 2021-05-19:
hi,
Do you solve it?
I have the same issue.

huziqi, 2021-05-20:
Hi,
I solved the problem in a wired way.
I changed the value of autoInitialize to 1 in the configure file. And it works
Best wishes and good luck~

huziqi, 2021-05-20:
i sovled the problem by changing the value of 'autoInitialize' to 1 in the configure file, and it works.

closed 2021-05-20 by huziqi

Abirami-Segar, 2021-06-04:
how did u set the rospath in the RosbagDataProvider.cpp to the location where you have your ros bag?

I mean this 

Rosbag Path: /home/hzq/dataset/euroc/MH_01_easy.bag

Any help would be much appreciated. Thank you.

umer1223, 2023-12-12:
... logging to /home/umer/.ros/log/6939e404-9936-11ee-b525-1f53059cc506/roslaunch-084-17380.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.18.11:36815/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/bow_batch_size: 1
 * /kimera_vio_ros/kimera_vio_ros_node/bow_skip_num: 1
 * /kimera_vio_ros/kimera_vio_ros_node/depth_cam_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/external_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/force_same_image_timestamp: True
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/image_transport: raw
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/odom_frame_id: odom
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/umer/catkin...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/publish_vlc_frames: True
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/robot_id: 0
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: 
 * /kimera_vio_ros/kimera_vio_ros_node/sensor_params_folder_path: 
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_external_odom: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd_registration_server: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_rviz: True
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/posegraph_viewer/frame_id: map
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [17394]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [17395]
[ INFO] [1702418197.561855348]: Initializing pose graph visualizer
ERROR: unknown command line flag 'colorlogtostderr'
ERROR: unknown command line flag 'log_prefix'
ERROR: unknown command line flag 'logtostderr'
ERROR: unknown command line flag 'minloglevel'
ERROR: unknown command line flag 'v'
================================================================================REQUIRED process [kimera_vio_ros/kimera_vio_ros_node-1] has died!
process has died [pid 17394, exit code 1, cmd /home/umer/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/umer/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/umer/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/umer/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/umer/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/umer/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --use_external_odometry=false --do_coarse_imu_camera_temporal_sync=false --do_fine_imu_camera_temporal_sync=false --lcd_no_optimize=false --lcd_no_detection=false --depth_image_mask= --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --minloglevel=0 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/umer/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --lcd_disable_stereo_match_depth_check=false --no_incremental_pose=false --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/umer/.ros/log/6939e404-9936-11ee-b525-1f53059cc506/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/umer/.ros/log/6939e404-9936-11ee-b525-1f53059cc506/kimera_vio_ros-kimera_vio_ros_node-1*.log
Initiating shutdown!
================================================================================
[kimera_vio_ros/posegraph_viewer-2] killing on exit
[kimera_vio_ros/kimera_vio_ros_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

can any one guide me, how to get rid of this error

avaniab, 2024-07-06:
Hi. I had the same error. Did you solve it? I tried setting autoInitialize to 1 but this didn't work.