MIT-SPARK/Kimera-VIO-ROS #142: Check failed: vio_params_.backend_params_->autoInitialize_ == 1 (0 vs. 1) Requested ground-truth initialization, but no gt_odom_topic was given.

zheng-xiao, 2021-05-19:
... logging to /home/zxn/.ros/log/5b69b4ae-b873-11eb-8d51-08beac02483f/roslaunch-zxn-pc-19376.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zxn-pc:46067/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/zxn/dev/kim...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: /home/zxn/dev/kim...
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /stereo_gray/dense_stereo/cam0/T_cam_imu: [[0.0140678333899...
 * /stereo_gray/dense_stereo/cam0/cam_overlaps: [1]
 * /stereo_gray/dense_stereo/cam0/camera_model: pinhole
 * /stereo_gray/dense_stereo/cam0/distortion_coeffs: [-0.0116026114041...
 * /stereo_gray/dense_stereo/cam0/distortion_model: equidistant
 * /stereo_gray/dense_stereo/cam0/flip_camera: True
 * /stereo_gray/dense_stereo/cam0/intrinsics: [461.341850302615...
 * /stereo_gray/dense_stereo/cam0/resolution: [752, 480]
 * /stereo_gray/dense_stereo/cam0/rostopic: /cam0/image_raw
 * /stereo_gray/dense_stereo/cam1/T_cam_imu: [[0.0118105423902...
 * /stereo_gray/dense_stereo/cam1/T_cn_cnm1: [[0.9999964092230...
 * /stereo_gray/dense_stereo/cam1/cam_overlaps: [0]
 * /stereo_gray/dense_stereo/cam1/camera_model: pinhole
 * /stereo_gray/dense_stereo/cam1/distortion_coeffs: [0.00573062561909...
 * /stereo_gray/dense_stereo/cam1/distortion_model: equidistant
 * /stereo_gray/dense_stereo/cam1/flip_camera: True
 * /stereo_gray/dense_stereo/cam1/intrinsics: [459.682043146586...
 * /stereo_gray/dense_stereo/cam1/resolution: [752, 480]
 * /stereo_gray/dense_stereo/cam1/rostopic: /cam1/image_raw
 * /stereo_gray/dense_stereo/depth/do_median_blur: False
 * /stereo_gray/dense_stereo/depth/use_mode_HH: True
 * /stereo_gray/dense_stereo/depth/use_sgbm: True
 * /stereo_gray/dense_stereo/first_camera_namespace: cam0
 * /stereo_gray/dense_stereo/first_output_frame: cam0
 * /stereo_gray/dense_stereo/input_camera_info_from_ros_params: True
 * /stereo_gray/dense_stereo/process_every_nth_frame: 1
 * /stereo_gray/dense_stereo/publish_tf: False
 * /stereo_gray/dense_stereo/scale: 1.0
 * /stereo_gray/dense_stereo/second_camera_namespace: cam1
 * /stereo_gray/dense_stereo/second_output_frame: cam1
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)
  /stereo_gray/
    dense_stereo (image_undistort/dense_stereo_node)

ROS_MASTER_URI=http://localhost:11311

process[stereo_gray/dense_stereo-1]: started with pid [19392]
process[kimera_vio_ros/kimera_vio_ros_node-2]: started with pid [19393]
process[kimera_vio_ros/posegraph_viewer-3]: started with pid [19394]
[ INFO] [1621415894.674323606]: Initializing pose graph visualizer
I0519 17:18:14.882470 19393 Logger.cpp:59] Opening output file: traj_gt.csv
I0519 17:18:14.883320 19393 RosBagDataProvider.cpp:52] Constructing RosbagDataProvider from path: 
 - Rosbag Path: /home/zxn/dev/kimera_ws/src/Kimera-VIO-ROS-master/V1_01_easy.bag
With ROS topics: 
 - Left cam: /cam0/image_raw
 - Right cam: /cam1/image_raw
 - IMU: /imu0
 - GT odom: 
I0519 17:18:14.884280 19393 RosBagDataProvider.cpp:83] Initialize Rosbag Data Provider.
I0519 17:18:14.884289 19393 RosBagDataProvider.cpp:201] Parsing rosbag data.
F0519 17:18:14.912034 19393 RosBagDataProvider.cpp:223] Check failed: vio_params_.backend_params_->autoInitialize_ == 1 (0 vs. 1) Requested ground-truth initialization, but no gt_odom_topic was given. Make sure you set ground_truth_odometry_rosbag_topic, or turn autoInitialize to false in BackendParameters.yaml.
*** Check failure stack trace: ***
    @     0x7fe66798eb0d  google::LogMessage::Fail()
    @     0x7fe6679909b1  google::LogMessage::SendToLog()
    @     0x7fe66798e63d  google::LogMessage::Flush()
    @     0x7fe667991389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fe666fd7708  VIO::RosbagDataProvider::parseRosbag()
    @     0x7fe666fd6456  VIO::RosbagDataProvider::initialize()
    @     0x7fe666fa721e  VIO::KimeraVioRos::createDataProvider()
    @     0x7fe666fa5c9a  VIO::KimeraVioRos::runKimeraVio()
    @     0x55c6a5c3dc21  main
    @     0x7fe66595fbf7  __libc_start_main
    @     0x55c6a5c3d8ba  _start
[kimera_vio_ros/kimera_vio_ros_node-2] process has died [pid 19393, exit code -6, cmd /home/zxn/dev/kimera_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/zxn/dev/kimera_ws/src/kimera-vio/vocabulary/ORBvoc.yml --flagfile=/home/zxn/dev/kimera_ws/src/kimera-vio/params/Euroc/flags/Mesher.flags --flagfile=/home/zxn/dev/kimera_ws/src/kimera-vio/params/Euroc/flags/VioBackend.flags --flagfile=/home/zxn/dev/kimera_ws/src/kimera-vio/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/zxn/dev/kimera_ws/src/kimera-vio/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/zxn/dev/kimera_ws/src/Kimera-VIO-ROS-master/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/zxn/.ros/log/5b69b4ae-b873-11eb-8d51-08beac02483f/kimera_vio_ros-kimera_vio_ros_node-2.log].
log file: /home/zxn/.ros/log/5b69b4ae-b873-11eb-8d51-08beac02483f/kimera_vio_ros-kimera_vio_ros_node-2*.log

Fatcat3Z, 2021-06-02:
Have you ever solve that? I meet the same problem. Hope you have a perfect solution!

Fatcat3Z, 2021-06-03:
I change the parameter 'autoInitialize' to 1, then the problem is sloved. However, nothing shows. The results are:
W0603 12:40:11.042245 19969 StereoMatcher.cpp:130] sparseStereoMatching: StereoFrame is already rectified!
W0603 12:40:11.042261 19969 StereoCamera.cpp:263] Rectifying already rectified stereo frame ...
E0603 12:40:11.062125 19971 Mesher.cpp:1729] Delaunay Triangle out of image (size: x: 0, y: 0, height: 480, width 752
 Triangle: x, y: 
327.379, 449.905
0, 2256
191.904, 425.463

BBlumi97, 2021-11-10:
I am facing the same issue. Did anyone solve it ?

wwj-53, 2023-07-16:
I am very sad that I have also encountered this problem. Has anyone solved it? Thank you.

LimHyungTae, 2024-08-20:
As mentioned in https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/136,

change `autoInitialize: 0` to `autoInitialize: 1` in the `Kimera-VIO/params/Euroc/BackendParams.yaml` file.