MIT-SPARK/Kimera-VIO-ROS #145: 'rosbag::BagIOException'   what():  Error opening file: ~/dataset/euroc/V1_01_easy.bag

Abirami-Segar, 2021-06-04:
**Description:**
Hi 

I am new to ROS and when i tried to launch the ros node, I got this error. Could someone help me?

THanks in advance!

**Command:**
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch

```

**Console output:**
```

# ... logging to /home/abirami/.ros/log/1c93a372-c51e-11eb-b2fd-f834415de486/roslaunch-abirami-4462.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://abirami:36805/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/abirami/cat...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: ~/dataset/euroc/V...
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [4479]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [4480]
[ INFO] [1622801894.567239603]: Initializing pose graph visualizer
I0604 12:18:18.610818  4479 Logger.cpp:59] Opening output file: traj_gt.csv
I0604 12:18:18.619415  4479 RosBagDataProvider.cpp:52] Constructing RosbagDataProvider from path: 
 - Rosbag Path: ~/dataset/euroc/V1_01_easy.bag
With ROS topics: 
 - Left cam: /cam0/image_raw
 - Right cam: /cam1/image_raw
 - IMU: /imu0
 - GT odom: 
I0604 12:18:18.626157  4479 RosBagDataProvider.cpp:83] Initialize Rosbag Data Provider.
I0604 12:18:18.626207  4479 RosBagDataProvider.cpp:201] Parsing rosbag data.
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error opening file: ~/dataset/euroc/V1_01_easy.bag
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 4479, exit code -6, cmd /home/abirami/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/abirami/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/abirami/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/abirami/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/abirami/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/abirami/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/abirami/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/abirami/.ros/log/1c93a372-c51e-11eb-b2fd-f834415de486/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/abirami/.ros/log/1c93a372-c51e-11eb-b2fd-f834415de486/kimera_vio_ros-kimera_vio_ros_node-1*.log

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

closed 2021-06-08 by Abirami-Segar

moksh019, 2021-11-12:
@Abirami-Segar  I'm facing same issue. Can you guide me if you have found the solution

Abirami-Segar, 2021-11-17:
Hi, make sure that you are feeding the right path to your bag file
location. Also make sure that in the program you have the correct location
to the bag file.

ZFDD96, 2021-11-24:
Hi, how do you solved this problem. I cannot understand your reply. Thanks.

ZFDD96, 2021-11-24:
> Hi, how do you solved this problem. I cannot understand your reply. Thanks.

oh, I get it. '~/' cannot be used!!!