MIT-SPARK/Kimera-VIO-ROS #146: 'rosbag::BagIOException'   what():  Error reading version line when launching kimera_vio_ros_euroc.launch

JoKoll, 2021-06-04:
**Description:**
I'm trying to get the example Euroc rosbag to run, but I haven't be able to run the kimera_vio_ros_euroc.launch file as the process dies with the following error:

terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error reading version line

However, I can launch the kimera_vio_ros.launch file with the euroc rosbag and have it run successfully on rviz, but I'm not sure if it's working completely correct. See kimera_vio_ros_screenshot image file.

[image]
**Command:**
Started roscore
In a new terminal, ran `roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch`
In a new terminal, ran `rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz`
In a new terminal, ran 'cd Documents/Euroc` then 'rosbag play --clock ./MH_01_easy.bag` This matches the location in the kimera_vio_ros_euroc.launch file for the rosbag.

rviz and the rosbag ran fine (see attached image), but the kimera wrapper is constantly failing. Launching the files in different orders did not change anything.
[image]
**Console output:**
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch
... logging to /home/blueksy/.ros/log/f59ab526-c586-11eb-94d7-d45d64bbac34/roslaunch-BlueSky-11141.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://BlueSky:36893/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/blueksy/cat...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: /home/blueksy/Doc...
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [11156]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [11157]
[ INFO] [1622846915.713321778]: Initializing pose graph visualizer
I0604 16:48:35.813683 11156 Logger.cpp:59] Opening output file: traj_gt.csv
I0604 16:48:35.815793 11156 RosBagDataProvider.cpp:52] Constructing RosbagDataProvider from path: 
 - Rosbag Path: /home/blueksy/Documents/Euroc/
With ROS topics: 
 - Left cam: /cam0/image_raw
 - Right cam: /cam1/image_raw
 - IMU: /imu0
 - GT odom: 
I0604 16:48:35.817874 11156 RosBagDataProvider.cpp:83] Initialize Rosbag Data Provider.
I0604 16:48:35.817889 11156 RosBagDataProvider.cpp:201] Parsing rosbag data.
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error reading version line
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 11156, exit code -6, cmd /home/blueksy/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/blueksy/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/blueksy/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/blueksy/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/blueksy/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/blueksy/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/blueksy/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/blueksy/.ros/log/f59ab526-c586-11eb-94d7-d45d64bbac34/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/blueksy/.ros/log/f59ab526-c586-11eb-94d7-d45d64bbac34/kimera_vio_ros-kimera_vio_ros_node-1*.log

```

*Rviz output:*
```
blueksy@BlueSky:~$ rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz
[ INFO] [1622848038.663461750]: rviz version 1.13.17
[ INFO] [1622848038.663541389]: compiled against Qt version 5.9.5
[ INFO] [1622848038.663568853]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1622848038.670750229]: Forcing OpenGl version 0.
[ INFO] [1622848039.307032764]: Stereo is NOT SUPPORTED
[ INFO] [1622848039.307084692]: OpenGL device: GeForce RTX 3090/PCIe/SSE2
[ INFO] [1622848039.307099880]: OpenGl version: 4.6 (GLSL 4.6).
```
**Additional files:**
Euroc rosbag file downloaded (MH_01_easy.bag) from [here](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets). It was completely downloaded before launching.

Please give also the following information:
* KimeraVIO branch, tag or commit used: d6bad104bc3a022a3e208f13ceb532d074e8c626
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Ubuntu 18.04.5 LTS
* Did you change the source code? (yes / no): no, other than adding the rosbag location

Abirami-Segar, 2021-06-05:
hi

I also have the same problem. If you find any solution pls share.

I think we are supposed to declare the rosbag file rather than just manually downloading them. I am too sure how to do it.

JoKoll, 2021-06-08:
I got this to work. In the "kimera_vio_ros_euroc.launch", the online argument default needs to be changed to "true". Should be around line 6.

  <arg name="online" default="true" />

closed 2021-06-08 by JoKoll