MIT-SPARK/Kimera-VIO-ROS #149: Delaunay triangle out of image.

Abirami-Segar, 2021-06-08:
**Description:**

**Command:**
```
I ran roscore in one terminal
I ran roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch in second terminal
i ran rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz in third terminal
finally I ran rosbag play --clock /PATH/TO/EUROC_ROSBAG

```

**Console output:**
```
E0608 13:05:32.386304 26390 Mesher.cpp:1729] Delaunay Triangle out of image (size: x: 0, y: 0, height: 480, width 752
 Triangle: x, y: 
0, 2256
-2256, -2256
127.905, 391.961
in the second terminal

Hit space to toggle paused, or 's' to step.
[ERROR] [1623150186.080917655, 1403715272.434165294]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
[ERROR] [1623150187.082348309, 1403715273.432094535]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
[ERROR] [1623150189.085481988, 1403715275.441065574]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
[ERROR] [1623150193.093245245, 1403715279.447157701]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
[ERROR] [1623150201.100508574, 1403715287.453445563]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
[ERROR] [1623150217.115947321, 1403715303.468024986]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
[ERROR] [1623150249.145071017, 1403715335.502617493]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
[ERROR] [1623150281.176692744, 1403715367.525855295]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
[ERROR] [1623150313.210660173, 1403715399.565383315]: Client [/gt_logger_node] wants topic /vicon/firefly_sbx/firefly_sbx to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [geometry_msgs/TransformStamped/b5764a33bfeb3588febc2682852579b0]. Dropping connection.
 [RUNNING]  Bag Time: 1403715418.878461   Duration: 147.161641 / 147.166583               
Done.
 
in the third terminal
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

banovers, 2022-11-16:
Have you solved the same problem for me

yusufersullu, 2025-01-04:
Is it solved? Same problem for me as well.