MIT-SPARK/Kimera-VIO-ROS #153: Zed2 Camera - No mesh being displayed in rviz

JoKoll, 2021-06-25:
**Description:**
I'm trying to run a Zed2 camera live with Kimera. I haven't been able to successfully display the mesh or PointCloud Time Horizon in rviz, but the feature tracks does work. I believe that Kimera-VIO-ROS has been installed correctly since I can run the Euroc rosbag example with the mesh and the PointCloud Time Horizon. I've also tried to run Kimera with a rosbag that I created, using the same topics and launch file (with the exception of sim_time = true) and was able to display the PointCloud Time Horizon in rviz, but not the mesh.

I have created custom param yaml files for the Zed2, and calibrated them with [Kalibr](https://github.com/ethz-asl/kalibr) as per the directions [here](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/hardware_setup.md). I've been using the grayscale imaging, as I couldn't calibrate the color images, and Kimera displays the following error when using the images_rect_color ros topic anyways: 

"F0624 18:40:36.960291 27317 RosDataProviderInterface.cpp:71] Check failed: cv_ptr->encoding == sensor_msgs::image_encodings::MONO8 (bgra8 vs. mono8) Expected image with MONO8, BGR8, or RGB8 encoding.Add in here more conversions if you wish."

**Command:**
Each line is in a new terminal:
```
roscore
roslaunch zed_wrapper zed2.launch
roslaunch kimera_vio_ros kimera_vio_zed2.launch
rviz
```
For rviz, I've just been manually setting the views when it opens and recently just checking to see if the mesh is being displayed after making a change.

**Console output:**
roscore
```
... logging to /home/blueksy/.ros/log/1a11799c-d54c-11eb-99b2-d45d64bbac34/roslaunch-BlueSky-26307.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://BlueSky:42307/
ros_comm version 1.14.11

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES

auto-starting new master
process[master]: started with pid [26344]
ROS_MASTER_URI=http://BlueSky:11311/

setting /run_id to 1a11799c-d54c-11eb-99b2-d45d64bbac34
process[rosout-1]: started with pid [26382]
started core service [/rosout]

```
Zed2 Launch
```
... logging to /home/blueksy/.ros/log/9472a604-d54a-11eb-99b2-d45d64bbac34/roslaunch-BlueSky-25813.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://BlueSky:46663/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.3
[... 634 lines cut ...]
```
I've identified two potential issues, but I'm not sure how to solve them and I'm not sure if they're related:
1. Both the "map_frame_id" and "world_frame_id" are set to "map" in the launch file. The Zed2 doesn't publish a "world" frame_id and when the "world_frame_id" is set to "world", there is warning in the TF of rviz since there is no transformation between the "world" frame and the "map" frame. I'm unsure how to rectify this or if it's even a problem. 

I know that the PointCloud Time Horizon is reliant on the "world" frame ID after I compared the Euroc rosbag to my Zed2 rosbag, where in the Zed2 rosbag, the frame_id in rostopic echo was listed as "map" as per the Kimera launch file. However, in both the live and rosbag scenarios, the mesh topic is not publishing. When being run live, running rostopic echo /kimera_vio_ros/mesh returns nothing, not even the message that nothing is being published. With the Zed2 rosbag, running rostopic echo /kimera_vio/ros/mesh returns that nothing is being published, despite the fact that /clock/ is running, similar to [Issue#78](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/78). I tried the suggested remedies in the comments, except changing the commits since I got the meshes to work with the Euroc rosbag, but still could not get the mesh to appear.

2. I repeatedly get the warning "W0624 18:16:44.947074 26136 Mesher.cpp:1377] Missing landmark information to build 3D Mesh." but I'm not sure how to fix it. It was mentioned in [Issue#126](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/126) as well, but there is no resolution to this issue. This error seems to disappear if Kimera/Zed2 live has been running for a bit. This warning does not appear when playing the Zed2 rosbag. 

I've also looked through [Issue#105](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/105), but nothing I changed showed the mesh in either the live Zed2 or the Zed2 rosbag.

This issue could also be completely unrelated to the above and I've just missed it. Any help to resolve this issue would be much appreciated.

Here is a screenshot of rviz when running kimera with the Zed2 livea and the rqt_graph:

[image]

[image]

Please give also the following information:
* KimeraVIO branch, tag or commit used: d6bad10
* GTSAM version used: 4
* OpenGV version used:
* OpenCV version used: 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04.5 LTS
* Did you change the source code? (yes / no): No, only created new param files and launch file.

JoKoll, 2021-06-25:
Following up on this, I think the problem may be related to the frame_ids, but not how I initially thought. I was doing some tests with a D435i and the major difference between them is that Kimera is publishing Frame IDs with the D435i but not with the Zed2. I even just copied the D435i launch file in the repository, and only changing the dataset value and the subscribed topics, but it made no difference. I was wondering what the purpose of this is:

```
<!-- Launch static TF node from base_link to velo_link -->
 <node pkg="tf" type="static_transform_publisher" name="velo_link_broadcaster"
   args="0 0 0 0 0 0 1 base_link velo_link 100"/>
```
When viewing the TFs in RVIZ with the D435i, it gives me complete transformations from the "world" frame id and "base_link", "map", "left_cam", and "right_cam" frame IDs. However, using a similar launch file with the Zed2, there are no transformations from the "world" frame id. Here's an image of the TFs in rviz of the Zed2:

[image]

With the D435i, even though the transformations from the "world" frame id to the id frames of the camera (as expected based on the text in the launch file stating that Kimera and Realsense publish conflicting Tfs), there are still the correct transformations to the "base_link", "left_cam" and "right_cam" frame ids. The frames in the visualization area also move when I move the camera, unlike with the Zed2 where nothing other than the "world" frame id shows up in the visualization area.

[image]

Since the _Launch static TF node from base_link to velo_link_ doesn't seem to work with the Zed2 as written, what's the background behind that code? I'm expecting that I'll have to make a custom transform in order to correctly associate Kimera with the Zed2 camera, but I don't know where "velo_link" is coming from. The [ros tutorials](http://wiki.ros.org/tf#static_transform_publisher) have some information on the static transform publisher, but not enough to know what I need to put there.

I think this missing transform is also why rostopic list can see the kimera_vio_ros topics (such as mesh, time_horizon_pointcloud, and odometry), but when I echo them, nothing is displayed (not even the message that they aren't publishing). 

Thanks again for any help on this topic.

MikeDmytrenko, 2022-07-27:
Any success with this issue? I`m having the same with ZED2i

vardhandongre, 2022-09-26:
Hi, I am having a similar issue with a zed2i camera, not sure what the cause of this is... some possible reasons I can think of are incorrect calibration or incorrect sensor frequency or the tfs, I might be wrong about this but in case someone resolved it please share how. Thanks!