MIT-SPARK/Kimera-VIO-ROS #158: installation getting stuck after occuration of 2 errors

G-Rizzle, 2021-09-20:
**Description:**
hello,

I'm trying to install the vio ros wrapper on my jetson. after the occuration of 2 fails (copy-error and missing boost library, see console output) it gets stuck.
I have no clue what could be wrong as i simply followed the given instructions for both the installation of ROS as well as the installation of the wrapper.

can somebody help me please?
thanks in advance

**Command:**
```
# replace this line with the command(s) you used
```

**Console output:**
```

racket@jetson2822:~/catkin_ws$ catkin init
Catkin workspace `/home/racket/catkin_ws` is already initialized. No action taken.
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/racket/catkin_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/racket/catkin_ws/build
Devel Space:        [exists] /home/racket/catkin_ws/devel
Install Space:      [unused] /home/racket/catkin_ws/install
Log Space:          [exists] /home/racket/catkin_ws/logs
Source Space:       [exists] /home/racket/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DGTSAM_TANGENT_PREINTEGRATION=OFF
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
racket@jetson2822:~/catkin_ws$ catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release -DGTSAM_TANGENT_PREINTEGRATION=OFF
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/racket/catkin_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/racket/catkin_ws/build
Devel Space:        [exists] /home/racket/catkin_ws/devel
Install Space:      [unused] /home/racket/catkin_ws/install
Log Space:          [exists] /home/racket/catkin_ws/logs
Source Space:       [exists] /home/racket/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DGTSAM_TANGENT_PREINTEGRATION=OFF
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
[... 180 lines cut ...]
  Boost.
Call Stack (most recent call first):
  CMakeLists.txt:11 (find_package)

cd /home/racket/catkin_ws/build/cv_bridge; catkin build --get-env cv_bridge | catkin env -si  /usr/bin/cmake /home/racket/catkin_ws/src/vision_opencv/cv_bridge --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/racket/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/racket/catkin_ws/install -DCMAKE_BUILD_TYPE=Release -DGTSAM_TANGENT_PREINTEGRATION=OFF; cd -
...............................................................................
Failed     << cv_bridge:cmake                              [ Exited with code 1 ]
Failed    <<< cv_bridge                                    [ 8.4 seconds ]     
Finished  <<< kimera_rviz_markers                          [ 17.5 seconds ]    
^A[build - 01:14.3] [6/19 complete] [8/8 jobs] [0 queued] [2 failed] [opencv3...[build - 01:48.0] [6/19 complete] [8/8 jobs] [0 queued] [2 failed] [opencv3...[b[build - 01:48.6] [6/19 ^C[build] Interrupted by user!                                                               
[build] Summary: 4 of 19 packages succeeded.                                               
[build]   Ignored:   1 packages were skipped or are blacklisted.                           
[build]   Warnings:  None.                                                                 
[build]   Abandoned: 13 packages were abandoned.                                           
[build]   Failed:    2 packages failed.                                                    
[build] Runtime: 2 minutes and 8.7 seconds total.

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  ubuntu 18.4
* Did you change the source code? (yes / no): no

HongLouyemeng, 2021-09-22:
> **Description:**
> hello,
>
> I'm trying to install the vio ros wrapper on my jetson. after the occuration of 2 fails (copy-error and missing boost library, see console output) it gets stuck.
> I have no [...]

Have you solved this problem? It has brought my research in this field to a standstill

G-Rizzle, 2021-09-22:
hi, particullary yes.

the boost issue is being solved by this:
https://github.com/ros-perception/vision_opencv/issues/341

simply comment out the "find_package(Boost REQUIRED python37)" statement. nevertheless there are a lot of other issues that i havent solved yet

HongLouyemeng, 2021-09-22:
I don't know why Kimera-Vio-Ros is so difficult to install.T_T---- Replied Message ***@***.***>Date09/22/2021 17:10 ***@***.***> ***@***.******@***.***>SubjectRe: [MIT-SPARK/Kimera-VIO-ROS] installation getting stuck after occuration of 2 errors (#158)
hi, particullary yes.
the boost issue is being solved by this:
ros-perception/vision_opencv#341
simply comment out the "find_package(Boost REQUIRED python37)" statement. nevertheless there are a lot of other issues that i havent solved yet

[
{
"target": "https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/158#issuecomment-924734814",
"url": "https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/158#issuecomment-924734814",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

HongLouyemeng, 2021-09-27:
> hi, particullary yes.
>
> the boost issue is being solved by this:
> [ros-perception/vision_opencv#341](https://github.com/ros-perception/vision_opencv/issues/341)
>
> simply comment out the [...]

Are you installing on the virtual machine

G-Rizzle, 2021-09-27:
> Are you installing on the virtual machine

no, I'm using the ROS-Wrapper

G-Rizzle, 2021-09-28:
I solved the problem. 

The issue was that the gflags file couldnt be copied, as I didnt have the permission in my computer gflag folder. I set this to permitted and now it works

closed 2021-09-28 by G-Rizzle

HongLouyemeng, 2021-09-28:
> 我解决了这个问题。
> 
> 问题是无法复制 gflags 文件，因为我在我的计算机 gflag 文件夹中没有权限。我将其设置为允许，现在可以使用了

Can you get 3D mesh in the rviz on Euro.bag?