MIT-SPARK/Kimera-VIO-ROS #165: KimeraVIO with D435i crashes everytime a small movement is made

matthewoots, 2021-11-02:
Thank you guys for the lovely VIO package, however I've ran into some roadblocks

I'm using a D435i on an NX and stream the image to my computer running on 18.04. However I have observed that VIO backend crashes everytime I make a slight movement, the error is `Backend did not return an output: shutting down Backend.` I'll see whether this occurs in bag files but for `live` it does not. Would like to also know what the error means, error is shown at `2.`

1. Realsense-ROS node : `roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation` Specific parameters are shown below in the console ouput
```
[ INFO] [1635828159.342970294]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1635828159.346006730]: infra1 stream is enabled - width: 640, height: 480, fps: 15, Format: Y8
[ INFO] [1635828159.347903670]: infra2 stream is enabled - width: 640, height: 480, fps: 15, Format: Y8
[ INFO] [1635828159.349855299]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1635828159.350176997]: gyro stream is enabled - fps: 200
[ INFO] [1635828159.350325702]: accel stream is enabled - fps: 63
[ INFO] [1635828159.350461287]: setupPublishers...
[ INFO] [1635828159.363782141]: Expected frequency for depth = 15.00000
[ INFO] [1635828159.595491955]: Expected frequency for infra1 = 15.00000
[ INFO] [1635828159.774085972]: Expected frequency for infra2 = 15.00000
[ INFO] [1635828159.951948303]: Expected frequency for color = 15.00000
[ INFO] [1635828160.108953444]: Start publisher IMU
[ INFO] [1635828160.126180659]: setupStreams...
[ INFO] [1635828160.152089562]: insert Depth to Stereo Module
[ INFO] [1635828160.152305435]: insert Color to RGB Camera
[ INFO] [1635828160.152439100]: insert Infrared to Stereo Module
[ INFO] [1635828160.152695998]: insert Infrared to Stereo Module
[ INFO] [1635828160.152882175]: insert Gyro to Motion Module
[ INFO] [1635828160.153161057]: insert Accel to Motion Module
 02/11 12:42:40,174 WARNING [547636277632] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
```
2 Kimera VIO : `roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch`, and after moving the device, this `error` shows up and closes the VIO
```
E1102 12:37:19.885910 21648 VioBackend.cpp:1251] Attempting to at the key "x118", which does not exist in the Values.
I1102 12:37:19.885977 21648 VioBackend.cpp:1613]  =============== START:CATCHING EXCEPTION =============== 
I1102 12:37:19.885982 21648 VioBackend.cpp:1640] Nr of factors in graph (smoother getFactors): 324, with factors:
I1102 12:37:19.885987 21648 VioBackend.cpp:1642] [
	Slot # 12: Point Prior: point key 	Factor  v135
	Slot # 16: Point Prior: point key 	Factor  v137
	Slot # 21: Point Prior: point key 	Factor  v138
	Slot # 23: Point Prior: point key 	Factor  v122
	Slot # 28: Point Prior: point key 	Factor  v123
	Slot # 33: Point Prior: point key 	Factor  v124
	Slot # 41: Point Prior: point key 	Factor  v126
	Slot # 43: Point Prior: point key 	Factor  v127
	Slot # 51: Point Prior: point key 	Factor  v129
	Slot # 53: Point Prior: point key 	Factor  v130
	Slot # 55: Point Prior: point key 	Factor  v132
	Slot # 56: Point Prior: point key 	Factor  v131
	Slot # 59: Point Prior: point key 	Factor  v134
	Slot # 63: Point Prior: point key 	Factor  v125
	Slot # 68: Point Prior: point key 	Factor  v128
	Slot # 73: Point Prior: point key 	Factor  v133
	Slot # 79: Point Prior: point key 	Factor  v136
	Slot # 81: Smart Factor (valid: yes, deg: no isCheir: no): 	Factor  x138 x139
	Slot # 82: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x138 x139
	Slot # 83: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x138 x139
	Slot # 84: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 85: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 86: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 87: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 88: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 89: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 90: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 91: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 92: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 93: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 94: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 95: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x137 x138 x139
	Slot # 96: Smart Factor (valid: yes, deg: no isCheir: no): 	Factor  x137 x138 x139
	Slot # 97: Smart Factor (valid: yes, deg: no isCheir: no): 	Factor  x137 x138 x139
	Slot # 98: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x136 x137 x138 x139
	Slot # 99: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x136 x137 x138 x139
	Slot # 100: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x136 x137 x138 x139
[... 457 lines cut ...]
	Slot # 232: Smart Factor (valid: yes, deg: no isCheir: no): 	Factor  x133 x134 x135 x136 x137 x138 x139
	Slot # 233: Smart Factor (valid: yes, deg: no isCheir: no): 	Factor  x134 x135 x136 x137 x138 x139
	Slot # 234: Smart Factor (valid: yes, deg: no isCheir: no): 	Factor  x134 x135 x136 x137 x138 x139
	Slot # 235: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x134 x135 x136 x137 x138 x139
	Slot # 236: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x135 x136 x137 x138 x139
	Slot # 237: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x135 x136 x137 x138 x139
	Slot # 238: Smart Factor (valid: NO!, deg: YES! isCheir: no): 	Factor  x135 x136 x137 x138 x139

I1102 12:37:19.891778 21648 VioBackend.cpp:1661]  ]
I1102 12:37:19.891780 21648 VioBackend.cpp:1664] Nr deleted slots: 0, with slots:
I1102 12:37:19.891782 21648 VioBackend.cpp:1666] [
	

I1102 12:37:19.891786 21648 VioBackend.cpp:1693]  ]
I1102 12:37:19.891788 21648 VioBackend.cpp:1696] Nr of values in state_ : 51, with keys:
[
	b122 b123 b124 b125 b126 b127 b128 b129 b130 b131 b132 b133 b134 b135 b136 b137 b138 v122 v123 v124 v125 v126 v127 v128 v129 v130 v131 v132 v133 v134 v135 v136 v137 v138 x122 x123 x124 x125 x126 x127 x128 x129 x130 x131 x132 x133 x134 x135 x136 x137 x138 
I1102 12:37:19.891827 21648 VioBackend.cpp:1702]  ]
I1102 12:37:19.891829 21648 VioBackend.cpp:1705] Nr values in new_values_ : 3, with keys:
[
	 b139  v139  x139 
I1102 12:37:19.891835 21648 VioBackend.cpp:1713]  ]
I1102 12:37:19.891837 21648 VioBackend.cpp:1724]  =============== END: CATCHING EXCEPTION =============== 
E1102 12:37:19.891871 21648 VioBackendModule.cpp:34] Backend did not return an output: shutting down Backend.
```

BBlumi97, 2021-11-08:
Hi, 
I am facing the same problem have you solved it yed ?

Regards, 
Benjamin

matthewoots, 2021-11-23:
@BBlumi97 I'm not sure whether it may be an issue with the NX not publishing the message to the correct rates, and dropping messages, found out when disabling `enable_sync` on the Realsense launch the rates are back to normal. Also when I use it on a better CPU, Kimera doesnt complain so probably you can try running on a desktop and longass usb cable extension for mapping or recording

BBlumi97, 2021-11-23:
@matthewoots Thanks for your reply! I will try it and report if only disabling the parameter will solve my problem.

BBlumi97, 2021-12-09:
sadly that didn't help me. The error just appears if the camera stands still and you start moving after a few seconds. If you start kimera while moving the error doesn't happen. Maybe @ToniRV can give us some insights in this error. 

Regards, 
Benjamin