MIT-SPARK/Kimera-VIO-ROS #166: /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/visualizer/PangolinDisplay.h:18:10: fatal error: pangolin/scene/axis.h: no such file or directory  #include <pangolin/scene/axis.h>           ^~~~~~~~~~~~~~~~~~~~~~~

shaoxh, 2021-11-06:
**Description:**
when build kimera-vio , a error message like it happened.

**Command:**
```
catkin build -j6
```

**Console output:**
```
Errors     << kimera_vio:make /home/shawn/Documents/catkin_ws_kimera/logs/kimera_vio/build.make.002.log
In file included from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/common/VioNavState.h:27:0,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:25,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/DataProviderInterface-definitions.cpp:14:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
 static void getValidKeypointsFromStatusKeypointsCV(
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp:17:0:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/EurocDataProvider.h: In constructor ‘VIO::EurocDataProvider::EurocDataProvider(const string&, const int&, const int&, const VIO::VioParams&)’:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/EurocDataProvider.h:220:11: warning: ‘VIO::EurocDataProvider::final_k_’ will be initialized after [-Wreorder]
   FrameId final_k_;    // end frame
           ^~~~~~~~
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/EurocDataProvider.h:205:13: warning:   ‘VIO::VioParams VIO::EurocDataProvider::vio_params_’ [-Wreorder]
   VioParams vio_params_;
             ^~~~~~~~~~~
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp:57:1: warning:   when initialized here [-Wreorder]
 EurocDataProvider::EurocDataProvider(const std::string& dataset_path,
 ^~~~~~~~~~~~~~~~~
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp: In member function ‘bool VIO::EurocDataProvider::parseImuData(const string&, const string&)’:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp:250:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (size_t i = 0u; i < gyr_acc_data.size() + 1u; i++) {
                         ~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/visualizer/DisplayFactory.h:24:0,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/playground/EurocPlayground.h:32,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/playground/EurocPlayground.cpp:1:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/visualizer/PangolinDisplay.h:18:10: fatal error: pangolin/scene/axis.h: 没有那个文件或目录
 #include <pangolin/scene/axis.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/kimera_vio.dir/src/playground/EurocPlayground.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp: In member function ‘const VIO::InitializationPerformance VIO::EurocDataProvider::getInitializationPerformance(const std::vector<long int>&, const std::vector<gtsam::Pose3>&, const VIO::VioNavState&, const Vector3&)’:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp:632:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 1; i < timestamps.size(); i++) {
                   ~~^~~~~~~~~~~~~~~~~~~
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp: In member function ‘VIO::KittiDataProvider::KittiData::operator bool() const’:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp:29:26: warning: suggest parentheses around comparison in operand of ‘==’ [-Wparentheses]
       timestamps_.size() == left_img_names_.size() == right_img_names_.size();
       ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:34:0,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/frontend/Camera.h:25,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
 static void getValidKeypointsFromStatusKeypointsCV(
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/common/VioNavState.h:27:0,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/EurocDataProvider.h:30,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp:17:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/common/vio_types.h: At global scope:
/home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
 static void getValidKeypointsFromStatusKeypointsCV(
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/common/VioNavState.h:27:0,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/backend/VioBackend-definitions.h:28,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/frontend/FrontendInputPacketBase.h:17,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderModule.h:29,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/MonoDataProviderModule.h:34,
                 from /home/shawn/Documents/catkin_ws_kimera/src/Kimera-VIO/include/kimera-vio/dataprovider/StereoDataProviderModule.h:27,
[... 845 lines cut ...]
//ADVANCED property for variable: gflags_DIR
gflags_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: glog_DIR
glog_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: gmock_build_tests
gmock_build_tests-ADVANCED:INTERNAL=1
//ADVANCED property for variable: gtest_build_samples
gtest_build_samples-ADVANCED:INTERNAL=1
//ADVANCED property for variable: gtest_build_tests
gtest_build_tests-ADVANCED:INTERNAL=1
//ADVANCED property for variable: gtest_disable_pthreads
gtest_disable_pthreads-ADVANCED:INTERNAL=1
//ADVANCED property for variable: gtest_force_shared_crt
gtest_force_shared_crt-ADVANCED:INTERNAL=1
//ADVANCED property for variable: gtest_hide_internal_symbols
gtest_hide_internal_symbols-ADVANCED:INTERNAL=1
targets_export_name:INTERNAL=GTestTargets

Please give also the following information:
* KimeraVIO branch, tag or commit used : commit 641576fd86bdecbd663b4db3cb068f49502f3a2c (HEAD -> master, origin/master, origin/HEAD)
* GTSAM version used: as project suggested. no change.
* OpenGV version used: as project suggested. no change.
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 18.04 
* Did you change the source code? (yes / no):  no

shaoxh, 2021-11-07:
Change Pangolin to another commit to solve it.

commit 86eb4975fc4fc8b5d92148c2e370045ae9bf9f5d (HEAD)
Author: Patrick Stotko <stotko@cs.uni-bonn.de>
Date:   Wed Oct 30 09:49:18 2019 +0100

Have no idea of it.

closed 2021-11-07 by shaoxh

HongLouyemeng, 2021-11-08:
> Change Pangolin to another commit to solve it.
> 
> commit 86eb4975fc4fc8b5d92148c2e370045ae9bf9f5d (HEAD) Author: Patrick Stotko [stotko@cs.uni-bonn.de](mailto:stotko@cs.uni-bonn.de) Date: Wed Oct 30 09:49:18 2019 +0100
> 
> Have no idea of it.

Hi. How did you solve this problem?

HongLouyemeng, 2021-11-08:
I get a error:defined reference to pangolin::Handler3D::Handler3D no define

shaoxh, 2021-11-10:
> I get a error:defined reference to pangolin::Handler3D::Handler3D no define

It seems a Handler3D function not exist in pangolin. In this situation, you could remove former installed pangolin and reinstall it. This commit version in my env works fine.