MIT-SPARK/Kimera-VIO-ROS #168: gflags error happens when build the code

DragonOnGit, 2021-11-17:
**Description:**
Hi developers, thanks for your work. I have a problem when I try to run `catkin build`, you can see it in the picture.
[image]

This happened when opencv3 is building in queue.

**Command:**
```
catkin build
```

**Console output:**
```

------------------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/nuc/Prometheus/devel:/opt/ros/melodic
Workspace:                   /home/nuc/catkin_ws
------------------------------------------------------------------------------------------
Build Space:        [exists] /home/nuc/catkin_ws/build
Devel Space:        [exists] /home/nuc/catkin_ws/devel
Install Space:      [unused] /home/nuc/catkin_ws/install
Log Space:          [exists] /home/nuc/catkin_ws/logs
Source Space:       [exists] /home/nuc/catkin_ws/src
DESTDIR:            [unused] None
------------------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DGTSAM_TANGENT_PREINTEGRATION=OFF
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------------------------------
[build] Found '16' packages in 0.0 seconds.                                                                                                                                                                
[build] Package table is up to date.                                                                                                                                                                       
Starting  >>> catkin_simple                                                                                                                                                                                
Starting  >>> disparity_image_proc                                                                                                                                                                         
Starting  >>> gtsam                                                                                                                                                                                        
Starting  >>> image_undistort                                                                                                                                                                              
Finished  <<< disparity_image_proc                         [ 0.1 seconds ]                                                                                                                                 
Finished  <<< catkin_simple                                [ 0.1 seconds ]                                                                                                                                 
Starting  >>> cmake_external_project_catkin                                                                                                                                                                
Starting  >>> gflags_catkin                                                                                                                                                                                
Starting  >>> kimera_rviz_markers                                                                                                                                                                          
Starting  >>> mesh_rviz_plugins                                                                                                                                                                            
Starting  >>> opencv3_catkin                                                                                                                                                                               
Starting  >>> opengv_catkin                                                                                                                                                                                
Starting  >>> pose_graph_tools                                                                                                                                                                             
Finished  <<< cmake_external_project_catkin                [ 0.1 seconds ]                                                                                                                                 
Finished  <<< opengv_catkin                                [ 0.2 seconds ]                                                                                                                                 
Finished  <<< gflags_catkin                                [ 0.2 seconds ]                                                                                                                                 
Finished  <<< pose_graph_tools                             [ 0.9 seconds ]                                                                                                                                 
Finished  <<< kimera_rviz_markers                          [ 1.0 seconds ]                                                                                                                                 
Finished  <<< image_undistort                              [ 1.3 seconds ]                                                                                                                                 
Starting  >>> glog_catkin                                                                                                                                                                                  
Finished  <<< mesh_rviz_plugins                            [ 1.2 seconds ]                                                                                                                                 
Finished  <<< glog_catkin                                  [ 0.1 seconds ]                                                                                                                                 
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << opencv3_catkin:make /home/nuc/catkin_ws/logs/opencv3_catkin/build.make.001.log                                                                                                               
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gflags/gflags-targets.cmake:37 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: gflags_shared;gflags_nothreads_shared
[... 8 lines cut ...]

make[2]: *** [opencv3_src-prefix/src/opencv3_src-stamp/opencv3_src-configure] Error 1
make[1]: *** [CMakeFiles/opencv3_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/nuc/catkin_ws/build/opencv3_catkin; catkin build --get-env opencv3_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << opencv3_catkin:make                          [ Exited with code 2 ]                                                                                                                          
Failed    <<< opencv3_catkin                               [ 1.5 seconds ]                                                                                                                                 
Abandoned <<< kimera_rpgo                                  [ Unrelated job failed ]                                                                                                                        
[... cut ...]
[build]   Abandoned: 4 packages were abandoned.                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 4.3 seconds total. 

```

Please give also the following information:
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 18.04
* Did you change the source code? (yes / no):  no

zgxsin, 2022-04-07:
Any update of this issue?

DragonOnGit, 2022-04-07:
> Any update of this issue?

Well, the main problem is opencv3 install position and version, if you have the same problem, just check these two parts. Maybe try to reinstall?

marcusabate, 2022-09-02:
Potentially related is [this issue](https://github.com/ethz-asl/eth_supermegabot/issues/28). Does this fix the problem?