MIT-SPARK/Kimera-VIO-ROS #170: Kimera node dies immediately on running euroc rosbag example

Kanishk598, 2022-03-30:
**Description:**
Whenever I am running the below command, kimera's ROS node is crashing. In the file `Kimera-VIO/params/Euroc/BackendParams.yaml` I changed `autoInitialize` from 0 to 1, and in the `kimera_vio_ros_euroc.launch` file line 28 I added the rosbag file path to the empty default field like this:
```
<arg name="rosbag_path" default="/home/vins/catkin_ws/src/V1_01_easy.bag" unless="$(arg online)"/>
```

**Command:**
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch
```

**Console output:**
```
... logging to /home/vins/.ros/log/bf5a1be0-b023-11ec-accc-94c691a56319/roslaunch-nuc18-7671.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nuc18:37517/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/vins/catkin...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: /home/vins/catkin...
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [7690]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [7691]
[ INFO] [1648643035.352824865]: Initializing pose graph visualizer
I0330 17:53:55.524214  7690 Logger.cpp:59] Opening output file: traj_gt.csv
I0330 17:53:55.525437  7690 RosBagDataProvider.cpp:52] Constructing RosbagDataProvider from path: 
 - Rosbag Path: /home/vins/catkin_ws/src/V1_01_easy.bag
With ROS topics: 
 - Left cam: /cam0/image_raw
 - Right cam: /cam1/image_raw
 - IMU: /imu0
 - GT odom: 
I0330 17:53:55.526459  7690 RosBagDataProvider.cpp:83] Initialize Rosbag Data Provider.
I0330 17:53:55.526469  7690 RosBagDataProvider.cpp:201] Parsing rosbag data.
I0330 17:53:56.653909  7690 RosBagDataProvider.cpp:324] Finished parsing rosbag data.
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 7690, exit code -11, cmd /home/vins/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/vins/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/vins/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/vins/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/vins/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/vins/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/vins/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/vins/.ros/log/bf5a1be0-b023-11ec-accc-94c691a56319/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/vins/.ros/log/bf5a1be0-b023-11ec-accc-94c691a56319/kimera_vio_ros-kimera_vio_ros_node-1*.log

```

Please give also the following information:
* KimeraVIO branch, tag or commit used: `641576fd86bdecbd663b4db3cb068f49502f3a2c`
* GTSAM version used: `0850e89b0e6148d16f00b7741d83b6325ada977d`
* OpenGV version used: `91f4b19c73450833a40e463ad3648aae80b3a7f3`
* OpenCV version used: `4.5.5`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): `Ubuntu 18.04`
* Did you change the source code? (yes / no): `Except parameter file and launch file, no (changes explained in the description above).`

Kanishk598, 2022-03-30:
I also observe that this error is most probably due to bad parsing of the [rosbag](http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/vicon_room1/V1_01_easy/V1_01_easy.bag) or bad rosbag itself. 
Whenever I am running `roslaunch kimera_vio_ros kimera_vio_ros.launch` it is waiting for the rosbag file, but as soon as I run the rosbag file in another console with --clock parameter, the console running kimera's ROS node dies:

```
[kimera_vio_ros/kimera_vio_ros_node-2] process has died [pid 8930, exit code -11, cmd /home/vins/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/vins/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/vins/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/vins/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/vins/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/vins/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/vins/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true left_cam/image_raw:=/cam0/image_raw right_cam/image_raw:=/cam1/image_raw imu:=/imu0 left_cam/camera_info:=/cam0/camera_info right_cam/camera_info:=/cam1/camera_info reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/vins/.ros/log/6ac129f2-b028-11ec-accc-94c691a56319/kimera_vio_ros-kimera_vio_ros_node-2.log].
log file: /home/vins/.ros/log/6ac129f2-b028-11ec-accc-94c691a56319/kimera_vio_ros-kimera_vio_ros_node-2*.log
```

hzx123q, 2022-04-10:
I have the same problem with you .

hzx123q, 2022-04-10:
The problem did't occur before.Maybe the software version's changes cause the problem.Could anyone can give some advice?

Kanishk598, 2022-04-11:
Which version did you test it on successfully? Can you please tell around which day you tested it successfully?

hzx123q, 2022-04-11:
I mean that the problem didn't occur in the previous issues.I haven't successfully run the kimera_vio_ros  yet.

hzx123q, 2022-04-17:
Try to contact the author to solve this problem？

minoru-kurata, 2022-04-18:
I have the same problem with you too. I don't know how to solve it.

esdandreu, 2022-04-20:
I tracked this issue and found that the program crashes when initializing a `gtsam::noiseModel::Diagonal`:
```
*no_motion_prior_noise = gtsam::noiseModel::Diagonal::Sigmas(sigmas);
```
[src/backend/VioBacked.cpp](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/src/backend/VioBackend.cpp#L1573)

Which is consistent with [a GTSAM issue](https://github.com/borglab/gtsam/issues/75)

shounak873, 2022-04-26:
I tried building with march native on, still getting segfault. Any suggestions?
`catkin config --cmake-args -DGTSAM_BUILD_WITH_MARCH_NATIVE=ON`

esdandreu, 2022-04-26:
I tried to build that way as well with no better luck. I've personally chosen to use another library for my project, one with better maintenance. I'm sorry I won't be able to help with this issue.

duda1202, 2022-04-27:
Which one are you using instead?

esdandreu, 2022-04-27:
Most probably I'm going to use one of the multiple forks of `ORB_SLAM2`, or `VINS_FUSION`. I haven't been able to try them yet.

duda1202, 2022-04-27:
I changed gtsam to 4.1.1 and it finally worked

canglangzhige, 2022-04-27:
> I changed gtsam to 4.1.1 and it finally worked

Could you tell me how can I change gtsam to 4.1.1.

duda1202, 2022-04-27:
I am not sure how you are doing the process. The easiest way is going inside gtsam and changing the version through git checkout so:

```
cd /path/to/gtsam
git checkout 4.1.1 
cd ~/catkin_ws
catkin build 
```

closed 2022-09-02 by marcusabate

iamsosadwwww, 2023-02-17:
> I tracked this issue and found that the program crashes when initializing a `gtsam::noiseModel::Diagonal`:
>
> ```
> *no_motion_prior_noise = gtsam::noiseModel::Diagonal::Sigmas(sigmas);
> ``` [...]

how can i track the issue ?