MIT-SPARK/Kimera-VIO-ROS #173: Launch file for a Realsense D435i

ShrutheeshIR, 2022-05-12:
**Description:**
Hello, this repo is fantastic. I have been able to get it running on the euroc dataset. However, I have trouble getting it running on live realsense D435i, with Kimera-VIO-ROS I tried running it on the default launch files of realsense, but I get no output from the kimera-vio module. Then if I change the default launch file of realsense with `enable_infra1:=true enable_infra2:=true`, I get an error : `Check failed: map_x_.size == img.size (480 x 640 vs. 480 x 848)`. I think that the realsense infrared image width is 848 perhaps, but I'm not very sure.

**Command:**
```
# replace this line with the command(s) you used
```
`roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation`, and then running `roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch`
No output

But with 
```roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_accel:=true enable_gyro:=true enable_imu:=true enable_infra1:=true enable_infra2:=true```
followed by 
```rosrun dynamic_reconfigure dynparam set /camera/stereo_module emitter_enabled 0```
and then 
```roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch```

Realsense Launch file
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="-1"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

[... 85 lines cut ...]
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>

      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
    </include>
  </group>
</launch>
```
**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used
* GTSAM version used - 4.1.0:
* OpenGV version used:
* OpenCV version used: 4.5.5-dev
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

duda1202, 2022-05-18:
Have you tried changing the infrared image width to 640? it should be at this line:
```
  <arg name="infra_width"         default="848"/>
```

marcusabate, 2022-09-02:
Closing as I confirmed locally that the widths need to match as @duda1202 suggested. Feel free to re-open if it still doesn't work.

closed 2022-09-02 by marcusabate