MIT-SPARK/Kimera-VIO-ROS #177: Kimera Node Dies as soon as Euroc rosbag is played

vardhandongre, 2022-09-01:
**Description:**
Kimera Nodes dies immediately as soon as the Euroc rosbag is played 

**Command:**
```
[Terminal 1] roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch online:=true
[Terminal 2] rosbag play --clock
```

**Console output:**
```

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/ijet/catkin...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

auto-starting new master
process[master]: started with pid [2261]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 89c704ec-2a0e-11ed-95aa-b53ec9ddbc69
process[rosout-1]: started with pid [2271]
started core service [/rosout]
process[kimera_vio_ros/kimera_vio_ros_node-2]: started with pid [2277]
process[kimera_vio_ros/posegraph_viewer-3]: started with pid [2279]
[ INFO] [1662047745.462406474]: Initializing pose graph visualizer
I0901 10:55:45.794559  2277 RosOnlineDataProvider.cpp:45] Waiting for ROS time to be valid... 
(Sim Time is enabled; run rosbag with --clock argument)
I0901 10:56:02.856274  2277 RosOnlineDataProvider.cpp:56] Requested initialization from ground-truth. Initializing ground-truth odometry one-shot subscriber.
W0901 10:56:02.858494  2277 RosOnlineDataProvider.cpp:65] Waiting for ground-truth pose to initialize VIO on ros topic: /kimera_vio_ros/gt_odom
E0901 10:56:05.859807  2277 RosOnlineDataProvider.cpp:82] Missing ground-truth pose while trying for 3.00369 seconds.
Enabling autoInitialize and continuing without ground-truth pose.
[kimera_vio_ros/kimera_vio_ros_node-2] process has died [pid 2277, exit code -11, cmd /home/ijet/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/ijet/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/ijet/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/ijet/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/ijet/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/ijet/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/ijet/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true left_cam/image_raw:=/cam0/image_raw right_cam/image_raw:=/cam1/image_raw imu:=/imu0 left_cam/camera_info:=/cam0/camera_info right_cam/camera_info:=/cam1/camera_info reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/ijet/.ros/log/89c704ec-2a0e-11ed-95aa-b53ec9ddbc69/kimera_vio_ros-kimera_vio_ros_node-2.log].
log file: /home/ijet/.ros/log/89c704ec-2a0e-11ed-95aa-b53ec9ddbc69/kimera_vio_ros-kimera_vio_ros_node-2*.log

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used: latest (9b8aeaedd407602b058ebcaa7aa1ed411a3abe9c)
* GTSAM version used: 4.1.1
* OpenGV version used:
* OpenCV version used: 4.2.0
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 20.04 + ROS Noetic
* Did you change the source code? (yes / no): No

vardhandongre, 2022-09-02:
Fixed it by changing the CMAKELIST.txt for mesh_rviz_plugin. It was a C++ version issue.

closed 2022-09-02 by vardhandongre

marcusabate, 2022-09-02:
For people in the future, here is the patch to apply in `mesh_rviz_plugins`, which will be merged on that branch soon:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index eb64d72..3de6a4e 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -5,11 +5,12 @@ find_package(catkin_simple REQUIRED)
 catkin_simple(ALL_DEPS_REQUIRED)
 
 set(CMAKE_MACOSX_RPATH 0)
-add_definitions(-std=c++11 -Wall)
+add_definitions(-std=c++14 -Wall)
 
 ## System dependencies are found with CMake's conventions
 find_package(PkgConfig REQUIRED)
 pkg_check_modules(EIGEN3 REQUIRED eigen3>=3.2)
+find_package(OpenCV REQUIRED)
 
 ############# RVIZ #######################
   pkg_check_modules(OGRE OGRE)
@@ -55,7 +56,8 @@ set(CMAKE_CXX_FLAGS_RELWITHDEBINFO "${CMAKE_CXX_FLAGS_RELWITHDEBINFO} ${SSE_FLAG
 ## Specify additional locations of header files
 ## Your package locations should be listed before other locations
 include_directories(include
-  ${EIGEN3_INCLUDE_DIRS})
+  ${EIGEN3_INCLUDE_DIRS}
+  ${OpenCV_INCLUDE_DIRS})
 
 set(MOC_FILES include/mesh_rviz_plugins/textured_mesh_display.h)
 cs_add_library(${PROJECT_NAME}
@@ -76,7 +78,8 @@ add_dependencies(${PROJECT_NAME} pcl_msgs_generate_messages_cpp)
 
 target_link_libraries(${PROJECT_NAME}
   ${OGRE_LIBRARIES}
-  ${QT_LIBRARIES})
+  ${QT_LIBRARIES}
+  ${OpenCV_LIBRARIES})
 
 install(
   FILES mesh_rviz_plugins.xml
```