MIT-SPARK/Kimera-VIO-ROS #180: [TIP] A successful installation of Kimera-VIO-ROS Guide

jaku-jaku, 2022-09-28:
##**Description:**
It has been a bit rabbit hole to get it working. 
I have just get it compiled properly in **Ubuntu 18.04 - Melodic** .

##**Guide**
### Step 1:
Follow the main guide https://github.com/MIT-SPARK/Kimera-VIO-ROS:
- [ ] https://github.com/MIT-SPARK/Kimera-VIO-ROS
- [ ] https://github.com/MIT-SPARK/Kimera-VIO-ROS

⚠️ Stop before `catkin build`, or `catkin clean` if you tried to built but in vain

### Step 2: (GTSAM)
As mentioned in https://github.com/MIT-SPARK/Kimera-VIO/issues/194, GTSAM has removed identity under `Pose3` and `Rot3`
1. A quick fix is to checkout the older version
```
$ cd catkin_ws/src/gtsam
$ git checkout c4184e192b4605303cc0b0d51129e470eb4b4ed1
```
2. modify Kimera-VIO to match latest GTSAM [TODO]
- [ ] Something to be implemented

### Step 3: (Kimera-VIO)
1. checkout the branch hydra support:
```
$ cd catkin_ws/src/Kimera-VIO
$ git checkout feature/hydra
```

### Step 4:
1. Download http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/vicon_room1/V1_01_easy/V1_01_easy.bag
2. in different terminals:
```
# Terminal 1:
$ roscore

# Terminal 2:
$ rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz

# Terminal 3:
$ roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch online:=true

# Terminal 4:
$ rosbag play --clock /PATH/TO/EUROC_ROSBAG
``` 

🥂 Good luck, folks!

Hopefully, this get maintained.

jaku-jaku, 2022-09-28:
According to https://github.com/MIT-SPARK/Kimera-VIO/pull/187, 
gtsam version commit: e5866799dff48239573cdd84964180867e50edd2

keiro23, 2022-10-13:
Hi, can you elaborate on what exactly I need to do in step 2.2(modify Kimera-VIO to match latest GTSAM [TODO])?

jaku-jaku, 2022-10-13:
> Hi, can you elaborate on what exactly I need to do in step 2.2(modify Kimera-VIO to match latest GTSAM [TODO])?

just replace functions whatever the compiler has thrown error on.