MIT-SPARK/Kimera-VIO-ROS #181: How can i save the real-time position estimation values ​​as a document?

keiro23, 2022-12-05:
Hello, I am an undergraduate student at university and a developer interested in SLAM. I am now running Kimera on ros using realsense d435i. Thanks to the interesting programs you have created, I am experimenting with various things in this program. Recently, I want to measure the accuracy of this program by outputting the estimated x, y, z values ​​and rotation values ​​as a document when experimenting with a realsense camera. When visualized with rviz, it is expected that Kimera is storing the above values ​​somewhere, but I am having a hard time not knowing exactly where and how. So, if you don't mind, I'd appreciate it if you could tell me the commands that can save the real-time position estimation values ​​as a document and how to add the commands to a file. Have a nice day today, and once again thank you for making such a wonderful program.

tianyilim, 2024-10-29:
Did you manage to find a way to do thsi?

tianyilim, 2024-10-30:
To follow up, one can take a look at the output logs folder.

You'd need to specify:
```xml
<arg name="log_output" default="true"/> <!-- change this here -->
<arg name="log_output_path"
  default="$(find kimera_vio_ros)/output_logs/$(arg dataset_name)"
  if="$(arg log_output)"/>
```

The trajectory will be saved under `log_output_path/traj_pgo.csv`.

Relevant lines of code: https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/d13b7b94cf4bc99bf9ed35b09163850463f4ed8d/launch/kimera_vio_ros_euroc.launch#L13C2-L16C32

keiro23, 2024-11-01:
Thanks for the answer! You will be blessed!