MIT-SPARK/Kimera-VIO-ROS #195: Mesher queue empty warning, poor mesh generation?

zippyzoo77, 2023-12-08:
Hello, I am trying to get the mesher working properly, and I keep getting very odd results that don't seem to line up with the environment, and I am wondering if that is due to the mesher queue issues that keep coming up? I put a piece of the outputs below so you can see, as well as a screenshot of what the mesh looks like. Any idea what is going wrong here and how to fix it to get a better mesh?

[image]

```
I1207 15:28:41.334957 932942 ThreadsafeQueue.h:305] Queue with id: mesher_frontend is getting full, size: 3
I1207 15:28:41.334964 932942 ThreadsafeQueue.h:305] Queue with id: lcd_frontend_queue is getting full, size: 3
I1207 15:28:41.334965 932942 ThreadsafeQueue.h:305] Queue with id: visualizer_frontend_queue is getting full, size: 3
W1207 15:28:41.366852 932864 Visualizer3DModule.cpp:133] Mesher queue is empty, yet Backend or Frontend queue is not!This should not happen since Mesher runs at Backend pace!
W1207 15:28:41.366900 932864 Visualizer3DModule.cpp:133] Mesher queue is empty, yet Backend or Frontend queue is not!This should not happen since Mesher runs at Backend pace!
I1207 15:28:41.366964 932864 KimeraVioRos.cpp:215] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Display [ms]                            	   61	0.616744	{0.704918 +- 2.59065}	[0,20]
Lcd [ms]                                	   31	0.313430	{69.0000 +- 4.26615}	[64,83]
Mesher [ms]                             	   31	0.313614	{3.61290 +- 1.68676}	[2,8]
Stereo Data Provider [ms]               	    0	
VioBackend [ms]                         	   31	0.313639	{4.22581 +- 2.33395}	[0,11]
VioFrontend Frame Rate [ms]             	 1452	14.7060	{2.91736 +- 0.529818}	[2,9]
VioFrontend Keyframe Rate [ms]          	   30	0.304269	{66.5333 +- 6.46334}	[53,90]
VioFrontend [ms]                        	 1483	14.9846	{4.27175 +- 12.8059}	[2,90]
Visualizer [ms]                         	   31	0.313427	{0.516129 +- 0.724383}	[0,3]
backend_input_queue Size [#]            	   31	0.312995	{1.00000 +- 0.00000}	[1,1]
data_provider_left_frame_queue Size [#] 	 1485	14.9738	{1.00000 +- 0.00000}	[1,1]
data_provider_right_frame_queue Size [#]	 1485	14.9738	{1.00067 +- 0.00000}	[1,2]
display_input_queue Size [#]            	   61	0.615418	{1.00000 +- 0.00000}	[1,1]
frontend_input_queue Size [#]           	 1483	14.9529	{1.00000 +- 0.00000}	[1,1]
lcd_backend_queue Size [#]              	   31	0.313532	{1.00000 +- 0.00000}	[1,1]
lcd_frontend_queue Size [#]             	 1483	14.9795	{36.3122 +- 22.7148}	[1,77]
mesher_backend Size [#]                 	   31	0.312982	{1.00000 +- 0.00000}	[1,1]
mesher_frontend Size [#]                	 1483	14.9532	{36.3122 +- 22.7148}	[1,77]
visualizer_backend_queue Size [#]       	   31	0.313626	{1.00000 +- 0.00000}	[1,1]
visualizer_frontend_queue Size [#]      	 1483	14.9835	{36.3122 +- 22.7148}	[1,77]
visualizer_lcd_queue Size [#]           	   31	0.313429	{1.00000 +- 0.00000}	[1,1]
visualizer_mesher_queue Size [#]        	   31	0.313613	{1.00000 +- 0.00000}	[1,1]
I1207 15:28:41.398730 932942 StereoVisionImuFrontend.cpp:113] ------------------- Processing frame k = 1485--------------------
I1207 15:28:41.398806 932942 StereoVisionImuFrontend.cpp:138] Current IMU Preintegration frequency: 192308 Hz. (52 us).
I1207 15:28:41.401396 932942 Tracker.cpp:149] Optical Flow Timing [ms]: 2
I1207 15:28:41.401603 932942 Frame.h:144] Frame id: 1485 at timestamp: 1701991721269879569
isKeyframe_: 0
nr keypoints_: 301
nr valid keypoints_: 301
nr landmarks_: 301
nr versors_: 301
size descriptors_: [0 x 0]
I1207 15:28:41.401628 932942 CameraParams.cpp:293]                                                            
******************** Camera Parameters ********************
Name                                                  Value
-----------------------------------------------------------
Camera ID ..........................................left_cam
Intrinsics: 
- fx....................................1071.58
- fy.................................................1072.37
- cu.................................................659.288
- cv.................................................399.247
frame_rate_: ....................................4.65661e-10
image_size_: 
- width...................................1280
- height.................................................720
depth_: 
- virtual_baseline.............................0.01
- depth_to_meters..........................................1
- min_depth................................................0
- max_depth...............................................10
- is_registered............................................1
-----------------------------------------------------------
I1207 15:28:41.401636 932942 CameraParams.cpp:294] - body_Pose_cam_: R: [
	0, 1.11022e-16, 1;
	-1, -2.22045e-16, 1.11022e-16;
	0, -1, 0
]
t: 000000 0.0375 000000
- K: [1071.578979492188, 0, 659.2884521484375;
 0, 1072.369140625, 399.2466430664062;
 0, 0, 1]
- Distortion Model:1
- Distortion Coeff:[0, 0, 0, 0, 0]
I1207 15:28:41.402297 932942 Tracker.cpp:368] Low mono disparity.
W1207 15:28:41.416780 932864 Visualizer3DModule.cpp:133] Mesher queue is empty, yet Backend or Frontend queue is not!This should not happen since Mesher runs at Backend pace!
I1207 15:28:41.436164 932942 FeatureDetector.cpp:208] Number of points detected : 766
I1207 15:28:41.436188 932942 FeatureDetector.cpp:222] Need n corners: 0
I1207 15:28:41.436221 932942 NonMaximumSuppression.cpp:102] Running Binning: 6
I1207 15:28:41.436228 932942 NonMaximumSuppression.cpp:119] Non Maximum Suppression Timing [ms]: 0
W1207 15:28:41.436231 932942 FeatureDetector.cpp:160] No corners extracted for frame with id: 1485
I1207 15:28:41.436241 932942 StereoMatcher.cpp:130] sparseStereoMatching: StereoFrame is already rectified!
I1207 15:28:41.436242 932942 StereoCamera.cpp:278] Rectifying already rectified stereo frame ...
I1207 15:28:41.459314 932942 StereoVisionImuFrontend.cpp:180] Keyframe 1485 with: 301 smart measurements
I1207 15:28:41.459421 932942 ThreadsafeQueue.h:305] Queue with id: mesher_frontend is getting full, size: 4
I1207 15:28:41.459427 932942 ThreadsafeQueue.h:305] Queue with id: lcd_frontend_queue is getting full, size: 4
I1207 15:28:41.459430 932942 ThreadsafeQueue.h:305] Queue with id: visualizer_frontend_queue is getting full, size: 4
I1207 15:28:41.459582 932943 RegularVioBackend.cpp:214] Tracker has a LOW_DISPARITY status.
I1207 15:28:41.463665 932943 VioBackend.cpp:1380] Backend: Update IMU Bias.
I1207 15:28:41.465530 932942 StereoVisionImuFrontend.cpp:113] ------------------- Processing frame k = 1486--------------------
I1207 15:28:41.465595 932942 StereoVisionImuFrontend.cpp:138] Current IMU Preintegration frequency: 217391 Hz. (46 us).
W1207 15:28:41.466771 932864 Visualizer3DModule.cpp:133] Mesher queue is empty, yet Backend or Frontend queue is not!This should not happen since Mesher runs at Backend pace!
I1207 15:28:41.468376 932942 Tracker.cpp:149] Optical Flow Timing [ms]: 2
I1207 15:28:41.468643 932942 Frame.h:144] Frame id: 1486 at timestamp: 1701991721336546086
isKeyframe_: 0
nr keypoints_: 301
nr valid keypoints_: 301
nr landmarks_: 301
nr versors_: 301
size descriptors_: [0 x 0]

```

yes0605, 2023-12-09:
我有一样的问题　
兄弟　你解决了吗

OmarZahra, 2023-12-19:
I am also having similar problems when working with the oak-d camera (not even getting meshes) which I believe are related. Can you please share the camera configuration yaml file, the params (in the Kimera_VIO resources), the launch file, and the rviz file. It would be nice to share as welll wish linux and ROS version you are using and gtsam version.

avaniab, 2024-07-13:
I have the same issue with Euros V1_01_easy.bag. Have you solved it?

yes0605, 2024-07-15:
我也没解决 我导给我安排别的活了 没有继续修改了

zippyzoo77, 2024-07-16:
I had a bit of luck when I moved into environments with more, flat horizontal and vertical surfaces, but I have since discovered that the odometry (and actually the mesh too, now I'm thinking of it) in the ros version is utter garbage, so I've been working with getting the non-ros version working with live data instead. However, I've had no response from the writers (or anyone else who has solved it) on either getting the ros odometry to not be garbage (see [here](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/204)) and or getting the non-ros working with live data (see [here](https://github.com/MIT-SPARK/Kimera-VIO/issues/227)). I've got a version of the non-ros that will compile and run the camera, but so far it just crashes right away after the camera boots.