MIT-SPARK/Kimera-VIO-ROS #198: Error during catkin build related to cephes/cephes_names.h: No such file or directory

Harshit0803, 2024-01-07:
**Description:**
After performing catkin build, there is an error in
/catkin_ws/devel/include/gtsam/3rdparty/cephes/cephes.h:4:10: fatal error: cephes/cephes_names.h: No such file or directory
    4 | #include "cephes/cephes_names.h"
    
  gtsam created an issue to fix it
  https://github.com/borglab/gtsam/pull/1704
  
  even after changing the files suggested in the pull request, still the error isn't solved

**Command:**
```
catkin build
```

**Console output:**
```

................................................................................................................................................................................
Finished  <<< gtsam                                        [ 4 minutes and 18.0 seconds ]                                                                                       
Starting  >>> kimera_rpgo                                                                                                                                                       
Starting  >>> opengv_catkin                                                                                                                                                     
________________________________________________________________________________________________________________________________________________________________________________
Errors     << kimera_rpgo:make /kimera_workspace/catkin_ws/logs/kimera_rpgo/build.make.000.log                                                                                  
In file included from /kimera_workspace/catkin_ws/devel/include/gtsam/nonlinear/internal/ChiSquaredInverse.h:24,
                 from /kimera_workspace/catkin_ws/devel/include/gtsam/nonlinear/GncOptimizer.h:31,
                 from /kimera_workspace/catkin_ws/src/Kimera-RPGO/include/KimeraRPGO/RobustSolver.h:17,
                 from /kimera_workspace/catkin_ws/src/Kimera-RPGO/src/RobustSolver.cpp:6:
/kimera_workspace/catkin_ws/devel/include/gtsam/3rdparty/cephes/cephes.h:4:10: fatal error: cephes/cephes_names.h: No such file or directory
    4 | #include "cephes/cephes_names.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/KimeraRPGO.dir/build.make:76: CMakeFiles/KimeraRPGO.dir/src/RobustSolver.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:220: CMakeFiles/KimeraRPGO.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /kimera_workspace/catkin_ws/build/kimera_rpgo; catkin build --get-env kimera_rpgo | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

................................................................................................................................................................................
Failed     << kimera_rpgo:make                             [ Exited with code 2 ]                                                                                               
Failed    <<< kimera_rpgo                                  [ 4.8 seconds ]                                                                                                      
Abandoned <<< kimera_vio                                   [ Unrelated job failed ]                                                                                             
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]                                                                                             
________________________________________________________________________________________________________________________________________________________________________________
Warnings   << opengv_catkin:make /kimera_workspace/catkin_ws/logs/opengv_catkin/build.make.000.log                                                                              
Cloning into 'opengv_src'...
Already on 'master'
Submodule 'python/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'python/pybind11'
Cloning into '/kimera_workspace/catkin_ws/build/opengv_catkin/opengv_src-prefix/src/opengv_src/python/pybind11'...
Submodule 'tools/clang' (https://github.com/wjakob/clang-cindex-python3) registered for path 'python/pybind11/tools/clang'
Cloning into '/kimera_workspace/catkin_ws/build/opengv_catkin/opengv_src-prefix/src/opengv_src/python/pybind11/tools/clang'...
make[3]: warning: -j8 forced in submake: resetting jobserver mode.
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
cd /kimera_workspace/catkin_ws/build/opengv_catkin; catkin build --get-env opengv_catkin | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

................................................................................................................................................................................
Finished  <<< opengv_catkin                                [ 1 minute and 11.4 seconds ]                                                                                        
[build] Summary: 11 of 14 packages succeeded.                                                                                                                                   
[build]   Ignored:   None.                                                                                                                                                      
[build]   Warnings:  7 packages succeeded with warnings.                                                                                                                        
[build]   Abandoned: 2 packages were abandoned.                                                                                                                                 
[build]   Failed:    1 packages failed.                                                                                                                                         
[build] Runtime: 5 minutes and 29.5 seconds total.  

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used: master
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`: 4.2.0
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Ubuntu 20.04
* Did you change the source code? (yes / no): No

ustccxy, 2024-01-12:
I'm having the same issue, how did you fix it?

ranjithlesoko, 2024-01-12:
same issue

kenscl, 2024-01-14:
For me the following worked:
1. cd into your_catkin_ws/src/gtsam
2. git checkout master  (apparently the dev branch is default)
3. git pull 
4. cd ..
5. catkin make