MIT-SPARK/Kimera-VIO-ROS #206: Kimera_VIO_ROS mono with euroc dataset is not working properly

skarthikbe96, 2024-06-19:
**Description:**
I have installed Kimera-VIO-ROS properly on Ubuntu 20.0 ROS1 noetic. Wen I am trying to run the euroc dataset with mono setting it is not able to publish any mesh or odometry trajectory. I have attached the screenshot of the rviz output

[image]

I need to test Kimera VIO using the Intel RealSense D435i camera, configured with the mono setting. Specifically, I will be using the /camera/color/image_raw topic for the image data.

**Command:**
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc_mono.launch 

rosbag play V1_01_easy.bag --clock

```

**Console output:**
```

roslaunch kimera_vio_ros kimera_vio_ros_euroc_mono.launch 
... logging to /home/student/.ros/log/59fa7fa6-2e5c-11ef-bb46-dbbb7155b57a/roslaunch-student-ThinkPad-L14-Gen-3-586742.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://student-ThinkPad-L14-Gen-3:36183/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/bow_batch_size: 1
 * /kimera_vio_ros/kimera_vio_ros_node/bow_skip_num: 1
 * /kimera_vio_ros/kimera_vio_ros_node/force_same_image_timestamp: True
 * /kimera_vio_ros/kimera_vio_ros_node/image_transport: raw
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/odom_frame_id: odom
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/student/sla...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/publish_vlc_frames: True
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/robot_id: 0
 * /kimera_vio_ros/kimera_vio_ros_node/sensor_params_folder_path: 
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_external_odom: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd_registration_server: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_rviz: True
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/posegraph_viewer/frame_id: map
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    rviz (rviz/rviz)
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [586767]
[... 70 lines cut ...]
VioBackend [ms]                           	        20	4.9	{1.4 +- 1.2}	[0.0, 4.0]	4.0
VioFrontend Frame Rate [ms]               	        59	15.3	{4.8 +- 2.2}	[1.0, 9.0]	4.0
VioFrontend Keyframe Rate [ms]            	        19	5.0	{24.1 +- 4.2}	[17.0, 31.0]20.0
VioFrontend [ms]                          	        79	19.2	{9.7 +- 8.8}	[1.0, 31.0]4.0
Visualizer [ms]                           	        20	4.9	{0.0 +- 0.0}	[0.0, 0.0]	0.0
backend_input_queue Size [#]              	        20	5.0	{1.0 +- 0.0}	[1.0, 1.0]	1.0
data_provider_left_frame_queue Size [#]   	        81	19.7	{1.0 +- 0.0}	[1.0, 1.0]	1.0
display_input_queue Size [#]              	        39	9.6	{1.0 +- 0.0}	[1.0, 1.0]	1.0
frontend_input_queue Size [#]             	        79	19.2	{1.0 +- 0.0}	[1.0, 1.0]	1.0
visualizer_backend_queue Size [#]         	        20	4.9	{1.0 +- 0.0}	[1.0, 1.0]	1.0
visualizer_frontend_queue Size [#]        	        20	5.0	{1.0 +- 0.0}	[1.0, 1.0]	1.0
^C[rviz-3] killing on exit

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used : 9274124d
* GTSAM version used: 4.2a9
* OpenGV version used: 
* OpenCV version used: 4.2.0
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu
* Did you change the source code? (yes / no): No

HesamTaherzadeh, 2024-08-03:
I also am facing the same issue, why is it asking for the groundtruth

tianyilim, 2024-10-29:
You need to edit the BackendParams.yaml file. This should be found in `~/catkin_ws/src/Kimera-VIO/params/Euroc/BackendParams.yaml`

Set `autoInitialize: 1`.

Then you don't need to provide a GT Odom source for initialization.