MIT-SPARK/Kimera-VIO-ROS #214: compile kimera-vio-ros:Error: GTSAM was built against a different version of Eigen

YingruiZhang, 2024-11-14:
**Description:**
I strictly follow the Kimera-VIO-ROS installation instructions. Everything goes well, but I got an error about gtsam was built against a different version of Eigen after running the catkin build. My laptop is Ubuntu 20.04 and also I tried rebuilding the gtsam from source and remove the gtsam from catkin workspace. However, it doesnot work. Could anyone help me with this issue? Thank you very much! 

**Command:**
```
# Compile code
catkin build
```

**Console output:**
```
Starting  >>> pose_graph_tools_ros                                             
Finished  <<< glog_catkin                                  [ 0.1 seconds ]     
Finished  <<< pose_graph_tools_ros                         [ 0.1 seconds ]     
_______________________________________________________________________________
Errors     << kimera_vio:make /home/yingrui/catkin_ws/logs/kimera_vio/build.make.004.log
In file included from /home/yingrui/catkin_ws/src/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:22,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/src/dataprovider/DataProviderInterface-definitions.cpp:14:
/home/yingrui/catkin_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^
In file included from /home/yingrui/catkin_ws/devel/include/gtsam/base/Matrix.h:28,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/Lie.h:26,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/VectorSpace.h:11,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Point2.h:20,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Cal3.h:24,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Cal3_S2.h:24,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/Camera.h:17,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/src/frontend/Camera.cpp:16:
/home/yingrui/catkin_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^
In file included from /home/yingrui/catkin_ws/devel/include/gtsam/base/Matrix.h:28,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/Lie.h:26,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/VectorSpace.h:11,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Point2.h:20,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Cal3.h:24,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Cal3DS2_Base.h:22,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Cal3DS2.h:23,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:17,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/src/frontend/CameraParams.cpp:16:
/home/yingrui/catkin_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^
In file included from /home/yingrui/catkin_ws/devel/include/gtsam/base/Matrix.h:28,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/BearingRange.h:21,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Pose3.h:22,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/OdometryParams.h:17,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/src/frontend/OdometryParams.cpp:15:
/home/yingrui/catkin_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^
In file included from /home/yingrui/catkin_ws/devel/include/gtsam/base/Matrix.h:28,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/Lie.h:26,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/base/VectorSpace.h:11,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Point2.h:20,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Cal3.h:24,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Cal3_S2.h:24,
                 from /home/yingrui/catkin_ws/devel/include/gtsam/geometry/Cal3_S2Stereo.h:20,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/StereoCamera.h:18,
                 from /home/yingrui/catkin_ws/src/Kimera-VIO/src/frontend/StereoCamera.cpp:15:
/home/yingrui/catkin_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* KimeraVIO branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

chengm0-0, 2024-11-20:
Hi, have you solved it?

NohSeungWook, 2024-11-27:
I solve this issue in the following way
1. reinstall GTSAM release 4.1.1
    git clone -b 4.1.1 git@github.com:borglab/gtsam.git
2. Unlike the installation guide, on Ubuntu 20.04 I set it up as follows.
    catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=ON -DGTSAM_TANGENT_PREINTEGRATION=OFF
3.build

I think it's GTSAM version matching issue. Maybe you can do without step 2.

closed 2024-12-11 by YingruiZhang